subsurface/subsurface #4365: MOD calculation calculated incorrectly in planner.

hhassey, 2024-11-07:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

In the dive planner, under the Available Gases The calculation of MOD (maximum operating depth) for _Deco Switch at_  is rounding up, creating in some instances warnings about high ppO2 on the planner result.

### Operating system:

Arch Linux 
### Subsurface version:

Subsurface 6.0.5214-CICD-release
### Steps to reproduce:

Go to the dive planner, under the available gases pane, insert a Tank and assign a mix of 67% Oxigen 0% Helium
Set the Deco P02 to 1.6 under the Gas Options pane.

### Current behavior:

 The planner automatically assigns a Deco Switch depth of 15m.
Calculating the depth manually,  [(1.6 / 0.67 )-1]*10 = 13.88m you would expect a deco switch depth of 14m  or even 13m if we round towards safety, but definitely  not 15 this throws a high ppo2 warning on the Dive plan Details. 

### Expected behavior:

Correct calculation of MOD and rounding down towards safety. 

### Additional information:

[image]

### Mentions:

glance-, 2024-11-07:
Your mod math is quite simplified, but we actually calculate it to  13608mm for the "default" surface pressure and water salt content.

By the same math a 100% will get a mod of 5809mm , which rounded for "safety" will end up being 3m which is just as "wrong".

hhassey, 2024-11-07:
Ok I agree with you, but I  never said to round up in 3m chunks!. 

But  to my point, if Subsurface is calculating the MOD  to 13608mm,  Why is it printing 15m as MOD and not 14m or 13 If "Safety is chosen"  ?

Regardless Subsurface is not being self consistent as it throws a Warning about high Ppo2  at the MOD actually calculated by it.

glance-, 2024-11-08:
3m or 10ft is the "industry standard" in deco stops so that's why it rounds to that.

Should we round mod to 3m/10ft in the planner? dunno. I'd guess we get something else equally odd somewhere else if we start rounding to 1m/1ft intervals.

If you don't like the mod, just change it.

atdotde, 2024-11-08:
IIRC there was also a problem when sticking too much to the rules, you could not with to EAN50 at 21m which of course many people will want to do. But at the moment I cannot conjure up the details.

But computing a gas switch depth and then complaining about it in the notes is a behaviour that could be improved....

glance-, 2024-11-08:
Rules? Rules?!? I see them more as guidelines.

atdotde, 2024-11-08:
Don't you have the underwater police in Sweden that arrests all divers at the 6m stop that have not obeyed the rules?

hhassey, 2024-11-08:
Switch depth does not necessarily need to happen at one of the deco stop depths ( in 3m chunks) there is actually a really nice check box called "Only switch at required stops"  that works perfectly in case you want to wait until your stop to switch gas. 

Respectfully I insist there is a bug. If as @glance- mentioned, Subsurface is calculating 13608mm as switch depth, but then print out 15m which is obviously deeper. (there is no rounding that can take you from 13.6 to 15m) so  anybody would expect a result of 14m which would be in agreement with other dive planners like Multideco which yields 14m for that 67% O2 mix

I am glad about the warning because otherwise this error could have passed unnoticed.

hhassey, 2024-11-15:
Just to add, information on this bug. 

If I set both the bottom PO2 and the deco PO2 to 1.6, you would expect the Deco Switch and the MOD calculation to be identical. Well they are not.

[image]

PR #4370 (2024-11-18): Round MOD to lower depths
When computing the suggested switch depth for a gas, we should take the next stop depth above the MOD, i.e. round down. Otherwise we can produce MOD violation warnings.

We need, however, a bit of fudge as otherwise we do not suggest to switch to o2 at 6m.

Fixes #4365

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

### Additional information:

### Releas [... cut]
closed 2024-11-18 by mikeller