subsurface/subsurface #4375: Code Smell Review of Dive.cpp (for Open Source Software Course)

FMain89, 2024-11-18:
### Describe the issue:

- [ ] Bug
- [X] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

In the dive.cpp file, several functions, such as per_cylinder_mean_depth(), have a high level of complexity and length, making the code difficult to read, maintain, and test. Reducing the length and complexity by breaking this function into smaller, modular helper functions can improve code clarity and maintainability.

### Operating system:

N/A (applies to the codebase)

### Subsurface version:

N/A (applies to the codebase)

### Steps to reproduce:

1. Open the dive.cpp file in the Subsurface repository.
2. Locate the per_cylinder_mean_depth() function.
3. Observe the function's length and complexity, including nested loops and conditionals.

### Current behavior:

The per_cylinder_mean_depth() function handles various aspects of per-cylinder mean depth calculations, including depth sampling, duration calculations, and handling different edge cases within a single block of code.

### Expected behavior:

The function should be broken down into smaller, more modular helper functions that handle distinct tasks such as sampling, duration computation, and validation, while maintaining the overall behavior of calculating per-cylinder mean depths.

### Additional information:

Full Disclosure: I am a Computer Science Student at Oregon State, taking an elective course in Open Source Software. One of the requirements of the course is to find a suitable project and contribute to it. I figured this would be a great opportunity to combine two passions of mine, diving and coding. 

Back to the project at hand:
Shorter functions improve readability, simplify debugging and maintenance, and enhance testability through isolated, unit-testable blocks.

### Mentions:

N/A

mikeller, 2024-11-18:
Thank you for your honesty.

Refactoring for improved maintainability is always a good thing, so I can't see a problem with contribution that implement this.
But to add a bit more context to this proposal, we are currently in a process of moving Subsurface from C to C++, so the expectation for refactoring will be that it leverages C++ wherever this makes sense.

@bstoeger how much will refactoring in `dive.cpp` interfere with any of your in-flight C++ conversion work?

Alternatively, if you are keen to contribute to Subsurface, you could have a look at our issue tracker at https://github.com/subsurface/subsurface/issues?q=is%3Aissue+is%3Aopen, and identify a bug or requested feature of suitable complexity for you to work on for your assignment.

bstoeger, 2024-11-19:
Frankly, I think `per_cylinder_mean_depth()`is far from the worst function in the code base.

> @bstoeger how much will refactoring in dive.cpp interfere with any of your in-flight C++ conversion work?

I'm not currently working on this file, so no worries. However, as I said, I think there are worse. Please be aware of our (idiosyncratic?) coding style(s).

Currently, I'm only working on the QtQuick-ification of the profile. However, I couldn't do anything in the last months owing to real-life obligations. And currently it doesn't look like this will improve soon.

closed 2025-05-24 by atdotde