subsurface/subsurface #4377: MacOS App crashes when trying to import media files (.jpg images)

chris-teichert, 2024-11-19:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

App crashes when trying to import media files into a dive

### Operating system:

macOS, Ventura 13.6.7

iMac 2017

### Subsurface version:

6.0.5276-CICD-release

Didn't test.

Official release

### Steps to reproduce:

1) open cloud storage
2) select dive
3) select "media" tab
4) right click in media tab and load media from file(s)
5) select 1 or more .jpg file
6) manually shift time of image, so it fits within the dive time
7) hit "OK" to start import

### Current behavior:

app crashes immediately after hitting "OK"

### Expected behavior:

app will not crash, but import the media files instead

### Additional information:

### Mentions:

chaehni, 2025-01-20:
I have the same problem. I now imported the media files on Windows and used the "find moved media files" option on Mac. Now I cannot even start Subsurface anymore. It crashes on startup.

mikeller, 2025-01-20:
I can't reproduce this here on MacOS 12.7.6 (the latest my ancient machine will run). Are you able to start Subsurface in a console (with `/Applications/Subsurface.app/Contents/MacOS/Subsurface -v -v -v`), and then post the output you are getting before it crashes?

chaehni, 2025-01-20:
Hi @mikeller 
Thanks for looking into this. I have the output but it doesn't really give an indication what fails. At least I can't tell.
Here you go:
```
INFO: validateGL(): created OpenGLContext.
INFO: validateGL(): obtained QOpenGLFunctions.
INFO: validateGL(): detected OpenGL version 2.1.
INFO: loading en-US translations
QObject::connect(QQuickWindow, QDeclarativeGeoMap_QML_6): invalid nullptr parameter
Populating font family aliases took 82 ms. Replace uses of missing font family "Bookman URW" with one that exists to avoid this cost. 
INFO: DivePixmaps DPR: 1,000000 metrics: 24 16 sz_bigger: 40
INFO: returning cloud URL https://ssrf-cloud-eu.subsurface-divelog.org/git/<email removed>[<email removed>]
INFO: loading dive data from: https://ssrf-cloud-eu.subsurface-divelog.org/git/<email removed>[<email removed>]
Cloud storage: checking connection to cloud server https://ssrf-cloud-eu.subsurface-divelog.org/git/<email removed>
INFO: Checking cloud connection...
INFO: IP used for cloud server access <ip removed>
2025-01-20 21:22:02.697 Subsurface[50627:22344819] +[IMKClient subclass]: chose IMKClient_Modern
2025-01-20 21:22:02.698 Subsurface[50627:22344819] +[IMKInputSession subclass]: chose IMKInputSession_Modern
Cloud storage: successfully checked connection to cloud server
INFO: git storage: accessing https://ssrf-cloud-eu.subsurface-divelog.org/git/<email removed>

INFO: git storage: Synchronising data file
INFO: git storage: update local repo

INFO: git storage: sync with remote https://ssrf-cloud-eu.subsurface-divelog.org/git/<email removed>[<email removed>]

INFO: git storage: Sync with cloud storage
INFO: git storage: delete proxy setting

Cloud storage: checking connection to cloud server https://ssrf-cloud-eu.subsurface-divelog.org/git/<email removed>
INFO: Checking cloud connection...
Cloud storage: successfully checked connection to cloud server
INFO: git storage: fetch remote https://ssrf-cloud-eu.subsurface-divelog.org/git/<email removed>

INFO: git storage: Successful cloud connection, fetch remote
INFO: git storage: certificate callback for host ssrf-cloud-eu.subsurface-divelog.org with validity 1

INFO: git storage: check remote status

INFO: git storage: try to update

INFO: git storage: Done syncing with cloud storage
INFO: git storage: Load dives from local cache
INFO: git load handling file 00-Subsurface

INFO: fingerprint 49ae1d69 00033f36 00000000 cbf9dea4 180c1b0d03

INFO: fingerprint 6b6ca4de a05b8b55 5276f205 90f74015 67867769

INFO: fingerprint e98ac5a0 67f4eca2 ab0fb04f 2ea326ca 03af4d66

INFO: git load handling file Site-091f551f

INFO: git load handling file Site-0ad39257

INFO: git load handling file Site-0b16c57f

INFO: git load handling file Site-0cb1959a

INFO: git load handling file Site-0cf26271

INFO: git load handling file Site-0f7c0450

INFO: git load handling file Site-12c2f9f5

INFO: git load handling file Site-15d53554

INFO: git load handling file Site-1663c289

INFO: git load handling file Site-193b8f6d
[... 2789 lines cut ...]
INFO: git load handling file +03=14=33

INFO: git load handling file +03=14=41

INFO: git load handling file +03=14=51

INFO: git load handling file +03=15=13

INFO: git load handling file +03=15=14

INFO: git load handling file +03=15=15

INFO: git load handling file +06=29=21

INFO: git load handling file +06=29=23

INFO: git load handling file +06=29=25

INFO: git load handling file +06=29=27

INFO: git storage: Successfully opened dive data
INFO: Profile calculation for dive 146 took 7 ms -- calculated ceiling preference is 1
INFO: Profile calculation for dive 146 took 9 ms -- calculated ceiling preference is 1

```

mikeller, 2025-01-21:
Thanks @chaehni - this is not giving much indeed. As a next step you could try running Subsurface in GDB (available on homebrew) and get a backtrace (`bt`) after the crash. If you download the macos build as an artifact on a pull request you'll get a debug build that will give line numbers in the stack trace.

chaehni, 2025-01-21:
Hi @mikeller 
I forgot to mention. I'm on Apple Silicon, therfore no GDB available. I used LLDB instead:

This is the output after running Subsurface with LLDB:
```
INFO: git storage: Successfully opened dive data
INFO: Profile calculation for dive 146 took 13 ms -- calculated ceiling preference is 1
INFO: Profile calculation for dive 146 took 15 ms -- calculated ceiling preference is 1
Process 68062 stopped
* thread #29, name = 'Thread (pooled)', stop reason = EXC_BAD_ACCESS (code=2, address=0x3061baff8)
    frame #0: 0x00007ff819166c2b libsystem_pthread.dylib`___chkstk_darwin + 55
libsystem_pthread.dylib`___chkstk_darwin:
->  0x7ff819166c2b <+55>: testq  %rcx, -0x8(%rcx)
    0x7ff819166c2f <+59>: cmpq   $0x1000, %rax  ; imm = 0x1000 
    0x7ff819166c35 <+65>: jb     0x7ff819166c51 ; <+93>
    0x7ff819166c37 <+67>: pushq  %rax
Target 0: (Subsurface) stopped.
```
This is the output from `bt`:
```
* thread #29, name = 'Thread (pooled)', stop reason = EXC_BAD_ACCESS (code=2, address=0x3061baff8)
  * frame #0: 0x00007ff819166c2b libsystem_pthread.dylib`___chkstk_darwin + 55
    frame #1: 0x000000000000009a
    frame #2: 0x000000010021441c Subsurface`Thumbnailer::fetchImage(QString const&, QString const&, bool) + 492
    frame #3: 0x0000000100214a82 Subsurface`Thumbnailer::getHashedImage(QString const&, bool) + 98
    frame #4: 0x00000001002169b0 Subsurface`Thumbnailer::processItem(QString, bool) + 80
    frame #5: 0x0000000100217fbc Subsurface`QtConcurrent::StoredFunctorCall0<void, Thumbnailer::fetchThumbnail(QString const&, bool)::$_1>::runFunctor() + 60
    frame #6: 0x00000001001589e0 Subsurface`QtConcurrent::RunFunctionTask<void>::run() + 48
    frame #7: 0x000000011529959c QtCore`___lldb_unnamed_symbol7166 + 156
    frame #8: 0x0000000115295207 QtCore`___lldb_unnamed_symbol7133 + 279
    frame #9: 0x00007ff81916b253 libsystem_pthread.dylib`_pthread_start + 99
    frame #10: 0x00007ff819166bef libsystem_pthread.dylib`thread_start + 15
````
This confirms that somehow images are the culprits. I hope you can make more sense of this than I can.

Thanks again for your help.

atdotde, 2025-01-21:
I can reproduce the crash here. It seems to me, in fetchImage it calls get_metadata which then crashes. 

Unfortunately, on this machine, I cannot get a meaningful debug session with breakpoints and single stepping to run. Could the problem be related to what was recently done in connection with libraw?

chaehni, 2025-01-21:
@atdotde 

This is my single step debug session starting from "get_metadata":

https://gist.github.com/chaehni/c59ab20771f110afd839d3becb06b1bd

mikeller, 2025-01-22:
@atdotde:

> I can reproduce the crash here. It seems to me, in fetchImage it calls get_metadata which then crashes.

Is this on a machine with Apple silicone, or Intel?

chaehni, 2025-01-22:
As pointed out by @atdotde the last change to `get_metadata` happened on 2024-09-03 in ae81b42fe2.
I installed Subsurface-6.0.5231-CICD-release.dmg from 2024-08-14 17:07 which is the last release before the change. This fixes my problem and now also the latest version of Subsurface runs. I assume the metadata is now stored and the crashing code is no longer run.

mikeller, 2025-01-22:
> As pointed out by @atdotde the last change to `get_metadata` happened on 2024-09-03 in [ae81b42](https://github.com/subsurface/subsurface/commit/ae81b42fe26fb0f93711d931513421aeb68cc3b5).
> I installed Subsurface-6.0.5231-CICD-release.dmg from 2024-08-14 17:07 which is the last release before the change.

The last release before ae81b42 is [5257](https://github.com/subsurface/nightly-builds/releases/tag/v6.0.5257-CICD-release), not 5231:

```
➜  subsurface git:(ae81b42fe) ✗ scripts/get-version.sh 
6.0.5258-patch.6.local%  
```

Are you able to test with this one, and [5258](https://github.com/subsurface/nightly-builds/releases/tag/v6.0.5258-CICD-release), which will allow us to confirm that it is the change in `get_metadata()` that is causing this.

atdotde, 2025-01-22:
5258 does not crash when adding a jpeg to a dive.

chaehni, 2025-01-22:
I did a binary search over the releases:
v6.0.5265 is the first release that crashes
v6.0.5264 I didn't test because there's no .dmg available
v6.0.5263 does not crash.

chaehni, 2025-01-22:
@mikeller Definitely weird. With older versions thumbnail generation works so this libraw code should not even be run. Also, it's interesting that it only fails on Mac.

PR #4437 (2025-02-09): Use thread safe version of libraw
We should use the thread safe version libraw_r since we are processing thumbnails in the background.

Fixes #4377 and thus supersedes PR #4424.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

### Additional information:

### Release note:

### Documentation change:
closed 2025-02-09 by bstoeger