subsurface/subsurface #4381: Cannot download from Scubapro's G2 via USB

p-neves, 2024-11-23:
### Describe the issue:
I am not able to download dives from my Scubapro G2 using the USB cable

- [X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When I try to use the USB cable to download dives from my Scubapro G2 I get an Error and no dives are downloaded.

### Operating system:
Arch Linux
Scubapro G2
No. I I've tried it in Windows using logTRAK and the download via USB works.

### Subsurface version:
Latest build. Compiled from master

Yes. The same happens using Subsurface-6.0.5285-CICD-release.AppImage

Self compiled

91d8bfef557d6c678c28147c2c76c06f645687b3
### Steps to reproduce:

1. Connect G2 to usb cable and press "log" button
2.  On Subsurface select "Import from dive computer"
3. Select Scubapro and G2 (vendor and Divecomputer);
4. Select downlod

### Current behavior:
I get an error: "Unable to open Scubapro (G2).
On the console I get:
INFO: Starting download from USBHID
INFO: downloading only new dives
[0.007134] ERROR: Failed to open the usb device (LIBUSB_ERROR_ACCESS). [in src/usbhid.c:597 (dc_usbhid_open)]
INFO: Finishing download thread: Unable to open  Scubapro (G2)

### Expected behavior:
The program should download the dives.

### Additional information:

### Mentions:

jefdriesen, 2024-11-25:
You don't have permission to access the USB device. The easiest solution is to install the libdivecomputer udev rules:
https://github.com/libdivecomputer/libdivecomputer/blob/master/contrib/udev/libdivecomputer.rules

(If Arch Linux doesn't have the plugdev group, or your user account isn't part of that group, you may need to modify the rule file.)

p-neves, 2024-11-25:
Thanks. I'll give it a try.

Cheers:
Pedro

mikeller, 2024-11-26:
The correct procedure is also documented in our official Subsurface manual:

https://subsurface-divelog.org/subsurface-user-manual/#S_HowFindDeviceName

p-neves, 2024-11-26:
@jefdriesen: your pointer did the trick. I thought I was already a member of the plugdev group but it turned out I wasn't.
Problem solved.
Thanks

closed 2024-11-26 by p-neves