subsurface/subsurface #4382: Subsbsurface for Windows freeze on start

MuadDib, 2024-11-24:
### Describe the issue: When openning Subsurface on windows it locks up and freezes with empty window

- [ x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When I try to open the app it freezes like this:
[image]
This is a log from Event Viewer:
[image]
I tried clearning the cache, running as admin and in compability mode. None helpped. Both %AppData%/Subsurface/ subsurface_err.log and subsurface_out.log have 0Kb

I did manage to get it to open by running the link from power shell:
[image]

### Operating system:
 OS Name	Microsoft Windows 11 Pro; Version	10.0.22631 Build 22631

### Subsurface version:
 
subsurface-6.0.5276-CICD-release-installer.exe
 
subsurface-6.0.5112-CICD-release-installer.exe
 

### Steps to reproduce:

Open the app... it freezes

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2024-11-25:
You are not sharing a lot of data.
So does opening it from PowerShell actually result in a working program, or just in some debug output?
Either way, can you send us the complete output (redirect to a file and attach the file to this issue, screenshots aren't all that useful for log data).
I assume you have successfully used it before and that you are opening a data file by default when starting Subsurface, correct?
What happens if you open Subsurface with a different data file (or an empty data file)?

mikeller, 2026-10-17:
Hi @MuadDib,

Thanks for reporting the freeze on start. To diagnose this we'd need some diagnostic output, running Subsurface with verbose logging enabled (or from a terminal) and sharing what gets printed before it freezes would help pin down where it's getting stuck.

If we don't hear back with that within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.