subsurface/subsurface #4386: Import via bluetooth fails on Mares Puck Pro +

janis1995, 2024-11-30:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I'm trying to import my dive log from a Mares Puck Pro + Dive Computer via bluetooth. The dive log contains 49 dives.
After importing 26 dives the import aborts and the dive computer ends up in its boot mode.
Reflashing firmware or removing battery is required for recovery of the dive computer.
If I try to connect and download after recovery again. Subsurface says no more new dives available on the computer.

### Operating system:

OS: Android 15 

Pixel 6A

### Subsurface version:

6.0.5276 (Official Release)
Installation via APK as the App can't be found in the Google Play Store for Android 15 devices.

### Steps to reproduce:

1) Pair the phone with the dive computer
2) Open the Subsurface App
3) Connect to the dive computer
4) Start downloading dives 

### Current behavior:

Download aborts after importing 26 dives and dive computer ends up in boot mode.

### Expected behavior:

Download finishes for all dives. It would be also nice to be able to select which dives to be imported.

### Additional information:

Mares Puck Pro + Firmware:  08.00.01

I would have attached the Log. But I can't find any app in android which is able to handle that much data from the clipboard.

### Mentions:

jefdriesen, 2024-12-05:
Subsurface mobile can generate an email with the log files as attachment, which you can send to yourself. The steps are explained here:
https://libdivecomputer.org/subsurface.html#mobile

janis1995, 2024-12-06:
Thanks for the hint this time even more dives got imported but still the dive, computer ends in boot mode.
[libdivecomputer.log](https://github.com/user-attachments/files/18039529/libdivecomputer.log)
[subsurface.log](https://github.com/user-attachments/files/18039530/subsurface.log)

jefdriesen, 2024-12-12:
Unfortunately the subsurface log is missing all the important info. You'll need to try with a build that includes this fix: https://github.com/subsurface/subsurface/pull/4385

janis1995, 2024-12-14:
I have retested it with version 6.0.5298!
[libdivecomputer.log](https://github.com/user-attachments/files/18135315/libdivecomputer.log)
[subsurface.log](https://github.com/user-attachments/files/18135316/subsurface.log)

janis1995, 2026-02-21:
Yes I assume firmware is crashing. Even I wrote to Mares, there reply was like just don't use subsurface for logging dives :/ 
So seems a bug in the puck pro+ I would say. Will close the issue!

closed 2026-02-21 by janis1995