subsurface/subsurface #4396: when replanning a dive, changes to GF aren't shown after save, but are saved

notionparallax, 2024-12-09:
### Describe the issue:

- [x] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

This is quite a niche situation I suspect. I was diving with my Suunto on gauge mode and using a borrowed Shearwater so that my algo matched the rest of the team. To save me having to pair the Shearwater with my phone, I used the data from the Suunto and modified it in the editor. I changed the GF, the altitude, and the water density. I'm guessing the default for GF is 30/75, and I modified it to be 50/75. In the image in the planner/editor (left) it's showing that correctly, but after pressing save, it goes back to 30/75.

If I re-edit the dive, it shows 50/75, so it is persisting somewhere. However, if I quit subsurface and reopen it, the dive is 30/75 in view, and edit mode.

|in the planner|after saving|
|--|--|
| [image] | [image] |

*Note:* I've used GF as the example throughout this report, but water density and altitude don't persist either.

### Operating system:

```
PS C:\Users\bdoherty> systeminfo /fo csv | ConvertFrom-Csv | select OS*, System*, Hotfix* | Format-List

OS Name             : Microsoft Windows 10 Pro
OS Version          : 10.0.19045 N/A Build 19045
OS Manufacturer     : Microsoft Corporation
OS Configuration    : Standalone Workstation
OS Build Type       : Multiprocessor Free
System Boot Time    : 5/12/2024, 11:32:49 AM
System Manufacturer : Microsoft Corporation
System Model        : Surface Laptop Go
System Type         : x64-based PC
System Directory    : C:\Windows\system32
System Locale       : en-gb;English (United Kingdom)
Hotfix(s)           : 39 Hotfix(s) Installed.,[01]: KB5045936,[02]: KB4534170,[03]: KB4537759,[04]: KB4542335,[05]:
                      KB4545706,[06]: KB4562830,[07]: KB4570334,[08]: KB4577586,[09]: KB5003791,[10]: KB5012170,[11]:
                      KB5015684,[12]: KB5020683,[13]: KB5026037,[14]: KB5046613,[15]: KB5006753,[16]: KB5011352,[17]:
                      KB5011651,[18]: KB5014032,[19]: KB5014035,[20]: KB5014671,[21]: KB5015895,[22]: KB5016705,[23]:
                      KB5018506,[24]: KB5020372,[25]: KB5022924,[26]: KB5023794,[27]: KB5025315,[28]: KB5026879,[29]:
                      KB5028318,[30]: KB5037018,[31]: KB5037240,[32]: KB5037995,[33]: KB5039336,[34]: KB5041579,[35]:
                      KB5043935,[36]: KB5043130,[37]: KB5046823,[38]: KB5005699,[39]: KB5048239
```
 I haven't tried this on other operating systems

### Subsurface version:
6.0.5231-CICD-release

### Steps to reproduce:

1) Take a dive that's recorded (I've tried it on a few dives, and it doesn't seem to matter if the computer is in gauge mode or not)
2) edit it in the planner
3) change the GF
4) save it
5) See if the dive shows the updated GF or not

### Current behavior:

As above

### Expected behavior:

I expected it to change the GF that is displayed to be 50/75 so that it persists across sessions.

### Additional information:

### Mentions:

notionparallax, 2024-12-09:
Hi @atdotde, I guess that one could record the GF used in the notes, or in the tags, but it seems a bit perverse to have the option to change the GF, and then not have that persist into the dive's log. 
The situations that I can see it being important is where: 
* I want to dive with a buddy who's got different GFs to me, and I change my computer to match theirs, then when I want to look at the dive at some point in the future, I wonder why my ascent looks weird.
* I am diving in differing conditions, e.g. it's very cold, or I'm working at depth, and I set a more conservative ascent in my computer, I'd want to record that in my log, and I wouldn't expect to need to write it in the notes (because that's the start of a slippery slope, why not just write everything in the notes if only some of the stuff gets recorded)
* I'm doing some experiments with how I feel at different ascent profiles, I want to be able to be able to focus on recording how I felt, and not need to keep track of the GFs too. (You might argue that this is a slippery slope the other way, why not have a field for _how did I feel after the dive_?)
* I've been diving for ages, and for years I used 99/99, and some new papers come out and I'm convinced to switch to 45/75, if I change my global setting at that point, it makes all my old dives look off.

I buy the argument for not changing the salinity, it's not all that obvious, but it's logical. However I'm not convinced that a one-size-fits-all approach to GFs is desirable. Generally, it feels like changes that the user makes should either persist, be blocked, or the UI should warn the user that the change isn't going to persist.

Admittedly, I don't know if GF (or deco algo at all) is transfered at all from the computer to Subsurface. If that was the case, the argument that it's all too hard would hold a lot more weight.

mikeller, 2024-12-10:
@notionparallax: I think there is a misunderstanding. @atdotde was only talking about how Subsurface is calculating its own gradient factor. Subsurface also supports reading the gradient factor, and related decompression information, from pretty much all dive computers supporting custom gradient factors.
You can normally find these gradient factors in the 'Extra Info' tab, and if you toggle on the 'DC reported ceiling' in the profile view, this ceiling will be based on the gradient factor used by the dive computer for the dive in question. This gradient factor will always be what was used by the dive computer on the actual dive.

notionparallax, 2024-12-10:
I guess I'm a bit lost then. I think the simplest way to express the bug report is that when I edit a dive, I would expect the changes to persist. If they're not going to persist for whatever reason, I would expect to be warned.

The more complicated bit that I'm confused about is about the global GF setting. If I go back to an old dive, and look in the extra information, it shows that the algo is RGBM, but at the top of the profile diagram it shows the global GF, rather than the DC algo. I can see the logic that the subsurface calculated ceiling is shown in green ink on the graph, and the subsurface global GF is shown in green text above. I don't actually have any dives with DC ceilings because I have always had it in gauge on any dives that I've gone past my NDL on. If I did go past my NDL, would that ceiling show in purple, with corresponding purple text?

I think what I'm dealing with in this area is confusion, rather than a bug, which might actually be more of a UX issue. However, I think that if a user changes something, and then that value is silently ignored, that's a bug.

dirkhh, 2024-12-10:
The Gradient Factor shown there is the one that Subsurface uses for its own Bühlmann calculation of the deco state during your dive.
Your dive computer may use a different algorithm (like RGBM), it may not tell us what algorithm it uses - it's different from vendor to vendor. We have no way to impact whatever the dive computer used during the dive, we have no way to simulate what the dive computer would do with different parameters.
The only thing we can do is do our own deco calculation. And that's not a per dive thing, but a global value for all dives.

Now that's the UI side of what you are shown when displaying the profile.

I think what you are telling us is that you expect the display for the GF used when planning a dive to match the GF used for that plan? I'm not really sure, actually...

mikeller, 2024-12-10:
Would this help with clarifying the UI:

[image]

mikeller, 2024-12-10:
@notionparallax:

> I can see value to having a global GF because it can show a sensible default for a dive with no set algo, but to have it override the known settings for a dive seems like a recipe for confusion to me.

It doesn't 'override' anything: You as the user are in full control to decide if you want to have the ceiling calculated by Subsurface shown, or the ceiling calculated by the dive computer, or both, or none:

[image]

[image]

(and these choices are persisted)

There is no standardised way for dive computers to tell Subsurface what decompression algorithm / settings were used - so 'known settings' is not something that exists, and there is no way for Subsurface to calculate anything based on such settings. Pretty much all we have ihn tangible information are the decompression stops as they were reported by the dive computer.

> @mikeller I think something like that would be good. I'm not sure if Subsurface GF is the right wording, because everything is Subsurface to the user, so it might not be clear that that's the globally set values?

That was a quick shot - any suggestions for improvements are welcome. 😊

notionparallax, 2024-12-10:
I think we've basically split this into two separate discussions. 

1) For the very weird situation where a dive is done with a computer that is recording depth but not a ceiling (like I was with my DC in gauge mode), I think I should be able to set a GF and for it to persist. That persistence should be (in my version of the world) in the settings, and in the visualisation. That might mean breaking a fundamental subsurface concept of there being one GF that subsurface reaches for whenever it's rendering a dive.

2) If a dive does have a GF stored, then the rendered ceiling, with all the nice compartments etc. it should use that GF when rendering. When I was saying override I mean only the GF used for rendering, not the user's choice to show the DC ceiling or not.

    Would it break things to allow the user to set a GF on a per dive basis? It might need a little bit more UI, but if the re-planner values were used, then it might just be a toggle between global and per-dive.

3) Bonus one, just to make sure we don't get totally sidetracked by this: if the user changes something in the planner, and it won't be saved when they click save, then they should be warned.

mikeller, 2024-12-10:
@notionparallax:

Re 1. and 2. - I think that this is not something that a lot of users will want to use - they will not be using their dive computer in a mode that intentionally discards information that might be relevant for their survival, and they will want to visualise the decompression information just as it was shown to them on the dive. But if you feel you would like to have this configurable then feel free to write the code to implement this, open a pull request, and then we can discuss.

Re 3., all of this is explained in depth in [the manual](https://subsurface-divelog.org/subsurface-user-manual/?lang=en#_non_recreational_open_circuit_dives_including_decompression) - it seems to be a bit moot to add warnings for things that work as described in the manual.

PR #4400 (2024-12-23): Profile: Clarify the Gradient Factor Display.
.

### Describe the pull request:

- [ ] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation
- [x] UI improvement

### Pull request long description:

Clarify that the gradient factor displayed in green in the profile is the gradient factor that is set in Subsurface, and not the one used by the dive computer

### Changes made:

### Related issues:

Fixes #4396.

### Additional information:

### Release note:

###
commit 39e47845 (2024-12-23): Profile: Clarify the Gradient Factor Display.

Clarify that the gradient factor displayed in green in the profile is
the gradient factor that is set in Subsurface, and not the one used by
the dive computer.

Fixes #4396.
closed 2024-12-23 by atdotde