subsurface/subsurface #4402: Import function with OCR for old fashioned paper log books

new2mc, 2024-12-26:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:
I am using a old fashioned paper dive log. I have now more than 100 logged dives. I tried subsurface and really like it. A feature would be so cool and handy to have an import function which can scan and edit the old fashioned log book (page by page) with OCR function and afterwards edit the read-in if necessary.

### Operating system:
Android 13 build T1RAS33.55-15-10-6
motorola edge 20 pro (XT2153-1)

### Subsurface version:
version 6.0.5298

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2024-12-26:
Indeed an intriguing idea. Except that the permutations (and the issues with reading handwriting) would make this almost certainly an unsolvable puzzle. The amount of effort that would go into this seems not remotely reasonable, considering the number of potential users.

closed 2024-12-26 by dirkhh

new2mc, 2024-12-27:
Ok, I thought such library or function\functionality already exists and it "only" has to be included and called...