subsurface/subsurface #4403: Digital license ID

new2mc, 2024-12-26:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:
I wasted to bring my license ID card (PADI OW) and wanted to go diving spontaneously though. I did not bring my license card unfortunately but I did not let my smart phone at home. It would be cool to have a function (maybe in personal settings) to save the license ID. (E.g. picture with camera or similar).

### Operating system:
Android 13 build T1RAS33.55-15-10-6
motorola edge 20 pro (XT2153-1)

### Subsurface version:
Version 6.0.5298

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2024-12-26:
Same answer as on #2107 -- just take a picture and store it on your phone.

closed 2024-12-26 by dirkhh

new2mc, 2024-12-27:
Point taken: keep it smart and simple! :) and will work without doubt!

Nevertheless, wouldn't it be nice to have feature to have it as let's say personal profile attribute at one spot?