subsurface/subsurface #4418: Gas mix only applied to one tank

matthewlai, 2025-01-13:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
With Shearwater Teric, I'm diving sidemount with the same gas in both tanks. On import, both pressures are shown, but only one is set to the mix I set in the computer. I am not doing gas switches on the computer as they have the same gas. The other one gets labeled as air. The gas bar does recognize that the whole dive is on EAN, so maybe we can just improve the heuristics to cover this situation?

### Operating system:
Android 11

### Subsurface version:
Subsurface-mobile 3.4.7(5.0.9.35) Android
Official release from Play Store.

### Steps to reproduce:
Dive with 2 transmitters and 1 gas set up to a EAN mix on Shearwater Teric.
Import into Subsurface-mobile through Bluetooth.

### Current behavior:
One tank gets labelled as AIR, the other the correct mix.

### Expected behavior:
Both tanks labelled with the correct mix

### Additional information:
I don't have the computer handy, but can get a log in a few days if necessary.

Thanks!

[image]

dirkhh, 2025-01-13:
Just a quick comment. 
The version in the play store is ancient and cannot be updated due to frustrating Google rules.
Current versions of Subsurface for Android can be found at https://subsurface-divelog.org/current-release

dirkhh, 2025-01-13:
I've talked about it a few times. It's the never ending treadmill, all designed to require an engineering team and a lawyer on the other end. Sign this document. Sign that document. Make this declaration. Make that declaration. Comply with these rules. With these other rules. You cannot use this API unless you do that. In order to do that you need to provide documentation why. Your documentation is insufficient. What is missing? No one knows, there's no way to ask.
Etc, etc, etc.
For an open source project with all volunteer work, that tries to access hardware (USB, storage, bluetooth, simply to be able to talk to dive computers and then store the data) it has gone from annoying to essentially a full time job. So about two years ago I gave up

matthewlai, 2025-01-13:
Ah I see yeah. That's probably too high above my pay grade to help with.

I have some Qt experience (Qt 4 way back in the days with C++ and have just picked up Qt 6 with Python now), so I may be able to help on the technical side once I have some free time again. I don't know much about mobile though.

mikeller, 2025-01-14:
> With Shearwater Teric, I'm diving sidemount with the same gas in both tanks. On import, both pressures are shown, but only one is set to the mix I set in the computer. I am not doing gas switches on [...]

This is a known shortcoming in Subsurface - what happens is that the dive computer does not care about about the link between gasmixes and tank sensors - it just sends two lists. And the 'heuristics' in Subsurface are to assume a 1 : 1 mapping between the entries in these lists.

While this may work sufficiently well for the > 95% of Subsurface users with a tank sensor who do single tank dives with a single sensor, it will result in a mismatch in most Tec diving setups with tank sensors, and definitely doesn't work for sidemount dives.

There are a number of threads where this is discussed, and risks and downsides for potential solutions are listed:
#4291 
https://groups.google.com/g/subsurface-divelog/c/dPCXd4aCpvY/m/TpTDSCYkAQAJ
https://groups.google.com/g/subsurface-divelog/c/aIIr0QJuxFY/m/NoyaZqnHAgAJ

But if you're keen to 'scratch your itch' and work on a solution for this I think this will be appreciated by many others - and I am happy to give guidance and feedback.

matthewlai, 2025-01-14:
@mikeller Thanks for the explanation!

Reading your posts linked, I guess a more reliable heuristic could be something like - for each tank, figure out the periods of time the pressure is decreasing, and look at the gas selected during those times to determine mix for the cylinder? This should cover single cylinder, doubles, doubles with and without transmitters on deco gas, and sidemount? CCR would be more tricky though.

I would love to work on this. I don't have time right at this moment but hopefully in a few weeks!

As a workaround for now, can I just define a redundant gas even if I never switch to it?

mikeller, 2025-01-15:
I like the heuristic approach - this could be done on the fly, which would allow us to retain the original information as it was downloaded from the dive computer, and store it in the currently used format that unfortunately lobs together potentially unrelated gasmix and tanksensor information.

There is also some information that we can get from the dive computer, like which tank sensors were used in a sidemount setup (if the user correctly configured them) - but not which gasmix they contained.

> As a workaround for now, can I just define a redundant gas on the computer even if I never switch to it?

As far as I am aware, Shearwater computers will not allow you to configure two gasmixes with identical gas.
But you can always change the 'air' that is added as the default gas for the second tank sensor to whatever gas you are using once you have downloaded the dive.

jefdriesen, 2025-01-21:
For the Shearwater dive computers, a recent libdivecomputer version will label the sidemount tanks with `DC_USAGE_SIDEMOUNT`. That still doesn't tell you which gas mix is present in those tanks, but it can serve as a hint that both tanks contain the same gas mix. For a dive with 2 (sidemount) transmitters and a single gas mix, that should result in the correct configuration.

closed 2026-10-17 by mikeller