subsurface/subsurface #4430: Infrared interface on newer linux?

jon011235, 2025-01-25:
Hello,
I want to use subsurface with my cressi donnatello, but it seems like linux has dropped IrDa support. Is anyone in the same situation?
How did you solve this? Does it simply work with a random IrDa USB Interface?

Thanks a lot in advance!

kruegerha, 2025-01-25:
I have a couple of old Aladin Tec2g, 3g and Primes, which I still use under linux (kernel 6.6, gentoo, as well as debian bookworm)
Infrared drivers have been dropped from the recent kernels, long ago.

However there drivers are still there, and being maintained! :
https://github.com/cschramm/irda
You can compile them as out-of-tree modules.

On top of that you'll need irda-utils (I use 0.9.18 with some patches applied).

I use an usb-IR-dongle which works with the stir4200 module ("hama"-branded).

To be honest, this is a quite a bit of a hassle, but yes it works, and no (unfortunately) not simply

Hannes

jon011235, 2025-01-27:
Thankds a lot!
I bought myself a dongle and will try it out!

How does the decision process for throwing out drivers work? IrDa does not seem to be *that* old and Linux is known to support things until really no one uses them anymore.

closed 2025-01-27 by jon011235

mikeller, 2025-01-27:
> How does the decision process for throwing out drivers work? IrDa does not seem to be _that_ old and Linux is known to support things until really no one uses them anymore.

See https://groups.google.com/g/subsurface-divelog/c/ZUfPrH7d9T0/m/JeXwW9WmDAAJ for some insights.

The reality is that IrDA simply tried to be everything to everybody - it did  serial, network, HID, ... And most of it never got enough traction to work reliably, yet it had the potential to sink the entire stack behind it.

It was a good protocol for dive computers, since they only ever used low bandwidth serial communications out of it. But apparently nobody put up their hand to create and maintain a simple userland serial IrDA implementation when support for the fully-fledged IrDA was yanked - probably because a that time everybody except for dive computer manufacturers had already moved on from IrDA to bluetooth, in order to create the same 'do-it-all' mess there...