subsurface/subsurface #4436: Error when building the app on macos (building libcurl)

sothawo, 2025-01-31:
Hello,

I am trying to build the app on macos (silicon), Sequoia 15.2, I like to add something to the stats functionality and want so see if I can get it running to make a PR then.

When running the build script at the point where libcurl should be built, I get

> configure: error: one or more libs available at link-time are not available run-time. Libs used at link-time: -lnghttp2 -lidn2 -lssl -lcrypto -lssl -lcrypto -lz

Some of these have been built into `src/install-root`:
````
> ls install-root/lib
total 15792
drwxr-xr-x  14 pjcc  staff   448B 31 Jan 21:32 .
drwxr-xr-x  14 pjcc  staff   448B 31 Jan 21:32 ..
drwxr-xr-x   4 pjcc  staff   128B 31 Jan 21:32 engines-1.1
-rwxr-xr-x   1 pjcc  staff   2.2M 31 Jan 21:32 libcrypto.1.1.dylib
-rw-r--r--   1 pjcc  staff   3.8M 31 Jan 21:32 libcrypto.a
lrwxr-xr-x   1 pjcc  staff    19B 31 Jan 21:32 libcrypto.dylib -> libcrypto.1.1.dylib
-rwxr-xr-x   1 pjcc  staff   524K 31 Jan 21:32 libssl.1.1.dylib
-rw-r--r--   1 pjcc  staff   774K 31 Jan 21:32 libssl.a
lrwxr-xr-x   1 pjcc  staff    16B 31 Jan 21:32 libssl.dylib -> libssl.1.1.dylib
-rwxr-xr-x   1 pjcc  staff   166K 31 Jan 21:32 libz.1.2.11.dylib
lrwxr-xr-x   1 pjcc  staff    17B 31 Jan 20:54 libz.1.dylib -> libz.1.2.11.dylib
-rw-r--r--   1 pjcc  staff   276K 31 Jan 21:32 libz.a
lrwxr-xr-x   1 pjcc  staff    12B 31 Jan 20:54 libz.dylib -> libz.1.dylib
drwxr-xr-x   5 pjcc  staff   160B 31 Jan 20:55 pkgconfig
````

Others like libnghttp2 or libidn2 I have in my system installed by homebrew, but they seem not to be found - and I think that this build should not rely on externally installed libraries.

What am I missing here? What information do you need in addition?

mikeller, 2025-01-31:
Hi @sothawo.

Sadly, I think the bit that you are missing is that building Subsurface on (and natively for) Apple silicon is not yet fully supported. There is an effort under way (#4300), but it's gotten stuck a while back.

The good news is that it's well possible that your build actually produces working binaries, even with the 'error' that is being logged.

sothawo, 2025-02-01:
The problem is that the build stops there with that error and doesn't build snything from the subsurface applicaiton itself.

sothawo, 2025-02-01:
Tries without `-build-deps`, had to installl libzip, but then I get the next failure.
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
HID_LIB
    linked by target "export-html" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface
    linked by target "Subsurface" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface
    linked by target "TestformatDiveGasString" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefTechnicalDetails" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefProxy" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefPartialPressureGas" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefGeocoding" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefMedia" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefDiveComputer" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefCloudStorage" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefUnits" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefGeneral" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestTagList" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefDivePlanner" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestPicture" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestRenumber" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefDisplay" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestGpsCoords" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefUpdateManager" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestPlan" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestParsePerformance" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestParse" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestHelper" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestMerge" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestUnitConversion" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestAirPressure" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestGitStorage" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefLanguage" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestProfile" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefEquipment" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQPrefLog" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestQML" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests
    linked by target "TestDiveSiteDuplication" in directory /Users/pjcc/development/3rdparty/subsurface/src/subsurface/tests

-- Generating done (1.0s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

I think I'll open an issue with the functionality I'd like to add, perhaps someone else can add it.

sothawo, 2025-02-01:
I still have a different laptop (intel macos) that likes to crash because of a faulty memory, but I'll try on that.

sothawo, 2025-02-01:
no luck with the intel mac either. I had to change github links to use ssh: instead of https: because of connection issues, but the build fails in the same way above on silicon with that 
> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
>Please set them or make sure they are set and tested correctly in the CMake files:
> HID_LIB

error.

sothawo, 2025-05-01:
Got Linux installed on my iMac now; no need to do Mac development for me, so I will close this issue.

closed 2025-05-01 by sothawo