subsurface/subsurface #4441: unexpected graphs of cylinder pressure in CCR dives

kruegerha, 2025-02-04:
in a CCR dive with three cylinders (diluent, oxygen, and one OC-gas; all without pressure sensors, but given start and end pressures)
the gas pressure plots show constant lines with a pressure equal to the given start pressure for oxygen and diluent.
I would expect a linear decrease for oxygen (and in a simple estimation the same for the diluent)

Things become strange when adding a limited period of bailout OC-usage (or even two).

Also, in a similar dive with two cylinders (DIL and bailout) with two limited bailout sessions, the first segments (start to first bailout) cylinder pressure plot, looks like the minimum pressure is not initialized..? and after the first bailout session, the diluents pressure is plotted as a constant line.
The pressure of the OC-cylinders during the (multiple) bailout session looks ok.

I attached a sample dive, which shows this effect.

This is the most recent master branch.

My guess is that somehow "segments" of diluent cylinders are not handled correctly along the 
populate_pressure_information
  fill_missing_tank_pressures
     fill_missing_segment_pressures
functions...  

I don't think that I'm able to track it down myself... any hints are welcome.

[CCR-gas.ssrf.gz](https://github.com/user-attachments/files/18663655/CCR-gas.ssrf.gz)

mikeller, 2025-02-04:
Yes, the handling of non-OC gas pressures in Subsurface currently is not what it should be.

Basically we make the assumption that, because there is no good way to predict the consumption of gases on a CCR (it certainly isn't strictly linear over time), no gas is consumed while on the loop.

Looking into this has been on my backlog for some time, but my way of dealing with it was just to not draw graphs for non-OC use of any gas (without sensors), as they don't represent realistic usage, and don't add to the information presented.

But happy to hear if you've got a better suggestion - and if you want to work on it go for it!

kruegerha, 2025-02-04:
Maybe it's better to completely skip the cylinder pressure plots for CCR segments then.

Where does this hugely negative   track[current].end   value in populate_pressure_information  come from?

mikeller, 2025-02-04:
> Maybe it's better to completely skip the cylinder pressure plots for CCR segments then.

Yeah, I think so.
If you combine the fact that gas consumption over time is low (O2) or even nil (diluent) with the fact that other things like depth changes, dil flushes, sensor checks, adherence to minimum loop can result in considerable temporary gas use, I suspect that any prediction of gas use over time will be wildly speculative. Graphing this will not add value.

If we wanted to show non-OC cylinder pressures for cylinders without a sensor, we could probably add them as dots at the point in the dive where they apply (i.e. start and end of use of the respective tank).

> Where does this hugely negative track[current].end value in populate_pressure_information come from?

I suspect that's the interpolation of your diluent start and end pressure gone wrong. The current interpolation algorithm is really only meant for OC, and will yield weird things for other use cases.