subsurface/subsurface #4442: unexpected behaviour with 2 gas on TERIC

Julaye, 2025-02-05:
Desktop version : 6.0.5299 
Mobile version : 6.0.5276

Have some dives with 2 gas : 
- 21% air on the principal, with a probe
- 50% O2 on the stage

For some reason, the TERIC is sorting the gas by % of O2, so stage is on position 1 and principal on position 2 (even if active first) in the gas list on the TERIC list of gas.

After the dive (without gas switch), seems the software is lost with a mixture. See the snapshots ! 

First snapshot : principal is now the 50% O2 ... but the bar values (start at 219 bar and finish at 62 bar with datas from the probe) are right !

[image]

Second snapshot, same issue in the editing window, principal is 50% O2 ... :

[image]

The same behaviour on mobile and desktop version.

PS: the same mix if I do a gas switch, the 02 information is not with the good tank ...

What do you need to investigate ?

mikeller, 2025-02-06:
> For some reason, the TERIC is sorting the gas by % of O2, so stage is on position 1 and principal on position 2 (even if active first) in the gas list on the TERIC list of gas.

Yes, this is the way Shearwater dive computers work, and there is no way to change it.

> After the dive (without gas switch), seems the software is lost with a mixture. See the snapshots !
>
> First snapshot : principal is now the 50% O2 ... but the bar values (start at 219 bar and finish [...]

Yes, this is normal as well - the dive computer does have no concept of which tank sensor belongs wo which gas, so the default it to associate the first tank sensor with the first gas in the list, and so on.

But if you go into the gas list on the Equipment tab, and right-click into a list heading, you can then check the 'Sensors' column to make it visible:

[image]

You can then click into the Sensors field for the gas that you want the sensor to be used for, and select the index of the column curently used for the sensor, and this will move the sensor to the desired gas:

[image]

[image]

At the moment the user experience for this is not super intuitive, I will have a look into making it a bit better.

Julaye, 2025-02-06:
Ok understood and It's working !

Thanks for you help.

closed 2025-02-06 by Julaye