subsurface/subsurface #4444: UDDF export: rating ininformationafterdive should  get exported if ratingvalue is empty

cheinelt, 2025-02-09:
In Mac version 6.0.5229-CICD-release the UDDF export exports empty ratingvalue inside of structures in informationafterdive. If a ratingvalue is empty it should not get exported and probably also rating itself should not get exported in this case.

It would be great if you could fix this. It let's UDDF parsers fail.

mturkia, 2025-02-12:
Apparently my original interpretation of UDDF specification was incorrect as the spec left room for interpretation. But running schema validation against Subsurface's UDDF export does produce quite a few errors. And the XML schema is the absolute truth, I suppose. So this needs to be fixed. (Schema is available in https://www.streit.cc/extern/uddf_v321/en/schema.html)

mikeller, 2025-02-17:
@cheinelt: Are you able to check if the artifacts on #4450 are fixing the problem for you?

PR #4450 (2025-02-22): Export: Fix 'rating' in UDDF Export.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Make the export of UDDF files compliant with the UDDF 3.2.3
specification for the `rating` element.

### Changes made:

### Related issues:

Fixes #4444.

### Additional information:

### Release note:

### Documentation change:
closed 2025-02-22 by bstoeger