subsurface/subsurface #4458: Different gases seen as air and plotted on the same line

OlivierV78, 2025-03-03:
Hi,

I still have my two sensors on my CCR (one on oxygen, the other on diluent) with different start and end pressures on this dive, but subsurface put them on the same plot.

The dive is in attachment.

Subsurface 6.0.53.46 ; Ubuntu linux

Thanks

[image]

[dive_ssrf.txt](https://github.com/user-attachments/files/19056233/dive_ssrf.txt)

mikeller, 2025-03-03:
Let me guess: You have used the 'Sensors' column to re-assign the tank sensors in this log?
After doing this, have you tried saving your dive, closing the logbook, then opening it again?

Currently, re-assigning sensors isn't working very well, and the profile and tank data is only updated properly when the dive is saved and then loaded again.
I am currently working on an improvement for this, but it's not quite ready yet.

OlivierV78, 2025-03-04:
Hi,

As there was a problem before I touched anything I had to try various manipulations.
But the problem was already present upon importation.

I closed it and reopened it. But even reopening today everything is the same.

Should I try deleting it and importing it cleanly again?

mikeller, 2025-03-04:
> As there was a problem before I touched anything I had to try various manipulations. But the problem was already present upon importation.

That's odd - the tank sensor data in the log file that you attached shows values for both sensors, but they are almost identical for pretty much all of the dive - almost as if the sensors were connected to the same tank, or the dive computer had a problem when receiving the sensor values.

I know there was a design fault with early Shearwater transmitters that could result in communication issues - maybe check if your transmitters happen to be affected by this: https://shearwater.com/blogs/community/voluntary-recall-notice-yellow-transmitters

> Should I try deleting it and importing it cleanly again?

Yes please. And if you could post the log directly as it comes off the computer this will help with further analysis.

OlivierV78, 2025-03-27:
@mikeller Everything seems ok after a new fresh clean import

OlivierV78, 2025-03-30:
The problem comes again on new dives after changing some information on the cylinder (but no changes to sensor).

He the two dives

[390_apres_modif.ssrf.txt](https://github.com/user-attachments/files/19525675/390_apres_modif.ssrf.txt)
[389_apres_modif.ssrf.txt](https://github.com/user-attachments/files/19525676/389_apres_modif.ssrf.txt)

mikeller, 2025-03-30:
> The problem comes again on new dives after changing some information on the cylinder (but no changes to sensor).

Which version of Subsurface were you using?

OlivierV78, 2025-03-30:
6.0.5367

I will try a new clean upload in the next days and hope to see which change do that.

mikeller, 2025-03-30:
@OlivierV78: Hard to tell what does this - can you please re-import the dives, and then send a logfile of how they look before you start editing them?

OlivierV78, 2025-04-12:
Hello @mikeller 
Did you see my previous post with clean imported dive and modified dive ?

OlivierV78, 2025-04-14:
@mikeller  Thank you for your reply and the time spent working on this beautiful software.

I don't remember exactly when the bug occurred.

I'll try to find the step that caused it, on this dive or others in the future if it happens again.

OlivierV78, 2025-04-16:
@mikeller 

Same problem on 2 new dives.
I'm not sure but it seems that the data are destroyed at opening or saving them with subsurface 6.0.5361 on windows.

[395_clean.ssrf.txt](https://github.com/user-attachments/files/19781315/395_clean.ssrf.txt)
[395_modified.ssrf.txt](https://github.com/user-attachments/files/19781316/395_modified.ssrf.txt)
[394_clean.ssrf.txt](https://github.com/user-attachments/files/19781318/394_clean.ssrf.txt)
[394_modified.ssrf.txt](https://github.com/user-attachments/files/19781317/394_modified.ssrf.txt)

OlivierV78, 2025-04-19:
Hello @mikeller 

I update to 6.0.5379 under windows and I found a very simple way to reproduce the bug :)
**it happens between saving to the cloud and opening it again.**

At this time I did not try a local save and a linux version.

Could you get the bug with this step by step ?

1° Open my 395_clean dive ; it is the imported file with no changes.

[395_clean.ssrf.txt](https://github.com/user-attachments/files/19820389/395_clean.ssrf.txt)

You will get this clean with no modification

[image]

2° Change the "use as" to "oxygen"  in the first gas line in this screenshot

[image]

The exported file is then 

[395_modif_oxygen.ssrf.txt](https://github.com/user-attachments/files/19820449/395_modif_oxygen.ssrf.txt)

3° Save to the cloud

4° Close and open subsurface with importing the cloud ; the gaz will be wrong and the two gas lines will be similar

[image]

%%%%%%%%%%%%

Version

[image]

mikeller, 2025-04-19:
@OlivierV78: Thanks, that's it, I am now able to reproduce the problem. It looks like there is something weird happening when storing in the cloud format, likely some legacy special treatment for 'oxygen' cylinders. I'll investigate.

OlivierV78, 2025-05-08:
Hi,
Any news about this bug ?

mikeller, 2025-05-08:
> Hi, Any news about this bug ?

No, I haven't had time yet to dig into this.

mikeller, 2025-05-21:
@Patrick-BS:

> I now have a sensor on my oxygen cylinder (only on this one), and when I read the data on Android, the tank sensor information is correctly shown in Subsurface. However, after opening the same dive log via cloud storage with the Windows client, the sensor data disappears eventually and seems to be removed from the log file.

I suspect that this might be due to the same problem that is fixed in #4523, which is yet to be merged. If you don't mind, could you please download and install the artifacts from this pull request and test with them?

OlivierV78, 2025-05-27:
Is it already integrated in the last nightly update ?