subsurface/subsurface #4460: Add support for Cressi BLE interfacee

2opremio, 2025-03-05:
I just purchased a [Cressi BT Interface for big screen computers](https://www.cressiusa.com/catalogue/cressitech.asp?id=1277):

[image] 

It's a BT-only interface, without the option to use a USB cable. It works great when syncing my Cressi Donatello (FW 1.02) with the Cressi app (both on iOS and on my mac).

However, Subsurface doesn't find the BT interface (I tried with both stable and nightly and in the same computer in which the Cressi app works fine):

[image] 

 Here's some info about the BT interface device, as obtained with the nrfConnect app:

[image]
[image]
[image]
[image]
[image]

mikeller, 2025-03-05:
Have you tried checking 'Show all BT devices' and then selecting the device?

2opremio, 2025-03-05:
I've tried now, but when manually choosing the device name (`4_1e52`) and I get an error:

[image]

Here is the log file:

```
Subsurface: v6.0.5347.0, built with libdivecomputer v0.9.0-devel-Subsurface-NG (6f19abf8d260dfb85257e33c08411dd08f9f4dc0)
[223.056111] INFO: Open: transport=32
[223.056120] INFO: Configure: baudrate=115200, databits=8, parity=0, stopbits=0, flowcontrol=0
[223.056122] INFO: Timeout: value=5000
[223.056123] INFO: RTS: value=0
[223.056124] INFO: DTR: value=0
[223.056125] INFO: Sleep: value=100
[223.156323] INFO: Purge: direction=3
[223.156403] INFO: Ioctl: request=0x40006203 (dir=1, type=98, nr=3, size=0)
[223.156406] INFO: Ioctl read: size=21, data=6E400003B5A3F393E0A9E50E24DC10B80000000000
[223.156408] ERROR: Failed to read the characteristic '6e400003-b5a3-f393-e0a9-e50e24dc10b8'. [in src/cressi_goa.c:503 (cressi_goa_device_foreach)]
```

2opremio, 2025-03-05:
I am on Sonorama 14.7.4

2opremio, 2025-03-05:
I've actually tried the same with another identical Donatello computer (the first one ran out of battery) with from Linux using Subsurface 6.0.5347-CICD-release and I am getting the same error:

```
Subsurface: v6.0.5347.0, built with libdivecomputer v0.9.0-devel-Subsurface-NG (6f19abf8d260dfb85257e33c08411dd08f9f4dc0)
[141.408898] INFO: Open: transport=32
[141.408917] INFO: Configure: baudrate=115200, databits=8, parity=0, stopbits=0, flowcontrol=0
[141.408920] INFO: Timeout: value=5000
[141.408923] INFO: RTS: value=0
[141.408925] INFO: DTR: value=0
[141.408927] INFO: Sleep: value=100
[141.509154] INFO: Purge: direction=3
[141.509184] INFO: Ioctl: request=0x40006203 (dir=1, type=98, nr=3, size=0)
[141.509187] INFO: Ioctl read: size=21, data=6E400003B5A3F393E0A9E50E24DC10B80000000000
[141.509193] ERROR: Failed to read the characteristic '6e400003-b5a3-f393-e0a9-e50e24dc10b8'. [in cressi_goa.c:502 (cressi_goa_device_foreach)]
```

If I connect to the device with `bluetoothctl`, these are the characteristics I get:

```
[bluetooth]# scan on
[bluetooth]# SetDiscoveryFilter success
[bluetooth]# Discovery started
[...]
[bluetooth]# [NEW] Device FD:59:C2:C4:6E:A1 4_254d
[...]
[bluetooth]# connect FD:59:C2:C4:6E:A1 
[...]
[CHG] Device FD:59:C2:C4:6E:A1 Connected: yes
[4_254d]# Connection successful
# info
Device FD:59:C2:C4:6E:A1 (random)
	Name: 4_254d
	Alias: 4_254d
	Paired: no
	Bonded: no
	Trusted: no
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Nordic Semiconductor AS.. (0000fe59-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (6e400001-b5a3-f393-e0a9-e50e24dc10b8)
	UUID: Nordic UART Service       (6e400001-b5a3-f393-e0a9-e50e24dcca9e)
	RSSI: 0xffffffce (-50)
	AdvertisingFlags:
  05                                               .               
	Battery Percentage: 0x5d (93)
```

2opremio, 2025-03-06:
CC @jefdriesen (since I see you are the author of most of the libcomputer goa computer code)

jefdriesen, 2025-03-06:
The reason for the failure is because the subsurface's BLE code doesn't implement the new `DC_IOCTL_BLE_CHARACTERISTIC_READ` ioctl yet.

Most BLE enabled dive computers use a communication protocol which simply emulates UART communication over the BLE link. There is a BLE service containing two characteristics: one for sending and one for receiving. The high level communication is done by sending commands and receiving the response over this virtual UART, just like with plain serial (or usb-serial) communication.

The Cressi dive computers are a bit different in this regard because they do have a UART service and characteristics, but they also have a few extra characteristics for information like the model number, firmware version and serial number. In their serial communication protocol, this information could be read by sending a command, but for the BLE communication they removed this command and we now have to read the info from those extra characteristics instead. Unfortunately this information is crucial, so we can't proceed without it.

The new ioctl's were added to libdivecomputer to support reading/writing such secondary characteristics.

2opremio, 2025-03-06:
Ah, I see, I didn’t know libdivecomputer delegated  the IO onto the consumer. The lack of error description didn’t help.

I can give a go to implement the ioctl (I assume that it will be a matter of mostly scaffolding since BLE support for the `normal` serial service should be there).

Do you have any tips for a newcomer?

2opremio, 2025-03-06:
Ah, it seems I just need to complete `qt_ble_ioctl`

2opremio, 2025-03-06:
I have bitten the bullet and implemented support for `DC_IOCTL_BLE_CHARACTERISTIC_READ` in #4463

Now subsurface can read the model/serial numbers, but reading the dives still fails:

Subsurface log:

```
INFO: Starting download from BLE
INFO: downloading only new dives
Using BlueZ LE DBus API
INFO: qt_ble_open(FD:59:C2:C4:6E:A1)
QLowEnergyControllerPrivateBluezDBus::connectToDevice()
QLowEnergyControllerPrivate setting state to QLowEnergyController::ConnectingState
###### "org.bluez.Device1" QMap(("Connected", QVariant(bool, true)))
###### "org.bluez.Device1" QMap(("ServicesResolved", QVariant(bool, true)))
QLowEnergyControllerPrivate setting state to QLowEnergyController::ConnectedState
INFO: connected to the controller for device FD:59:C2:C4:6E:A1
QLowEnergyControllerPrivate setting state to QLowEnergyController::DiscoveringState
Dedicated Battery1 service available
Using battery service via GattService1 interface
QLowEnergyControllerPrivate setting state to QLowEnergyController::DiscoveredState
INFO: finished service discovery, start discovering characteristics
INFO: Found service {00001801-0000-1000-8000-00805f9b34fb}
INFO:  .. ignoring standard service
INFO: Found service {0000180a-0000-1000-8000-00805f9b34fb}
INFO:  .. ignoring standard service
INFO: Found service {0000180f-0000-1000-8000-00805f9b34fb}
INFO:  .. ignoring standard service
INFO: Found service {0000fe59-0000-1000-8000-00805f9b34fb}
INFO:  .. ignoring standard service
INFO: Found service {6e400001-b5a3-f393-e0a9-e50e24dc10b8}
INFO: starting service characteristics discovery
INFO:    .. service state changed to 2
INFO: Found service {6e400001-b5a3-f393-e0a9-e50e24dcca9e}
INFO:  .. recognized service Nordic Semi UART
INFO: starting service characteristics discovery
INFO:    .. service state changed to 2
INFO:  .. done discovering services
Read Desc: "\x00\x00"
Read Char: "{6e400003-b5a3-f393-e0a9-e50e24dc10b8}" "4d25000004"
Read Char: "{6e400004-b5a3-f393-e0a9-e50e24dc10b8}" "6700"
Read Char: "{6e400005-b5a3-f393-e0a9-e50e24dc10b8}" "0100"
INFO:    .. service state changed to 3
Read Char: "{6e400003-b5a3-f393-e0a9-e50e24dcca9e}" "00"
Read Desc: "\x01\x00"
INFO:    .. service state changed to 3
INFO: Found service {6e400001-b5a3-f393-e0a9-e50e24dcca9e} Unknown Service
INFO:    c: {6e400002-b5a3-f393-e0a9-e50e24dcca9e}
INFO:    c: {6e400003-b5a3-f393-e0a9-e50e24dcca9e}
INFO:         d: {00002902-0000-1000-8000-00805f9b34fb}
INFO: Using service {6e400001-b5a3-f393-e0a9-e50e24dcca9e} as preferred service
INFO:  .. enabling notifications
INFO: Using read characteristic {6e400003-b5a3-f393-e0a9-e50e24dcca9e}
INFO: now writing "0x0100" to the descriptor {00002902-0000-1000-8000-00805f9b34fb}
Init CCC change to "0100" "{6e400003-b5a3-f393-e0a9-e50e24dcca9e}" "{6e400001-b5a3-f393-e0a9-e50e24dcca9e}"
Write Desc: "{00002902-0000-1000-8000-00805f9b34fb}" "0100"
INFO: BLE write completed
INFO: Reading BLE characteristic {6e400003-b5a3-f393-e0a9-e50e24dc10b8} from service {6e400001-b5a3-f393-e0a9-e50e24dc10b8}
INFO: Obtained characteristic value: 4d25000004
INFO: Reading BLE characteristic {6e400004-b5a3-f393-e0a9-e50e24dc10b8} from service {6e400001-b5a3-f393-e0a9-e50e24dc10b8}
INFO: Obtained characteristic value: 6700
INFO: Reading BLE characteristic {6e400005-b5a3-f393-e0a9-e50e24dc10b8} from service {6e400001-b5a3-f393-e0a9-e50e24dc10b8}
INFO: Obtained characteristic value: 0100
INFO: 16:05:57 packet SEND 0200
INFO: 16:05:57 packet WAIT
Read Char: "{6e400003-b5a3-f393-e0a9-e50e24dc10b8}" "4d25000004"
Read Char: "{6e400004-b5a3-f393-e0a9-e50e24dc10b8}" "6700"
Read Char: "{6e400005-b5a3-f393-e0a9-e50e24dc10b8}" "0100"
INFO: Deleting BLE object
INFO:    .. service state changed to 0
QLowEnergyControllerPrivate setting state to QLowEnergyController::ClosingState
Low Energy Controller deleted while connected.
INFO: Finishing download thread: Dive data import error
```

libdivecomputer log:

```
Subsurface: v6.0.5347.0, built with libdivecomputer v0.9.0-devel-Subsurface-NG (6f19abf8d260dfb85257e33c08411dd08f9f4dc0)
[1758.775305] INFO: Open: transport=32
[1758.775327] INFO: Configure: baudrate=115200, databits=8, parity=0, stopbits=0, flowcontrol=0
[1758.775330] INFO: Timeout: value=5000
[1758.775332] INFO: RTS: value=0
[1758.775334] INFO: DTR: value=0
[1758.775363] INFO: Sleep: value=100
[1758.875600] INFO: Purge: direction=3
[1758.875641] INFO: Ioctl: request=0x40006203 (dir=1, type=98, nr=3, size=0)
[1758.880305] INFO: Ioctl read: size=21, data=6E400003B5A3F393E0A9E50E24DC10B84D25000004
[1758.880320] INFO: Ioctl: request=0x40006203 (dir=1, type=98, nr=3, size=0)
[1758.880389] INFO: Ioctl read: size=18, data=6E400004B5A3F393E0A9E50E24DC10B86700
[1758.880392] INFO: Ioctl: request=0x40006203 (dir=1, type=98, nr=3, size=0)
[1758.880418] INFO: Ioctl read: size=18, data=6E400005B5A3F393E0A9E50E24DC10B80100
Event: vendor=4D2500000467000100
Event: model=4 (0x00000004), firmware=103 (0x00000067), serial=9549 (0x0000254d)
[1758.880607] INFO: Sleep: value=2000
[1760.881021] INFO: Write: size=2, data=0200
[1765.883595] INFO: Read: size=0, data=
[1765.883608] ERROR: Failed to receive the answer. [in src/cressi_goa.c:241 (cressi_goa_device_download)]
[1765.883615] ERROR: Failed to read the logbook data. [in src/cressi_goa.c:591 (cressi_goa_device_foreach)]
```

mikeller, 2025-03-06:
@2opremio: Since you are already working on this (and thanks for that by the way), you could also add the device information for the new device into https://github.com/subsurface/subsurface/blob/master/core/btdiscovery.cpp - this should be straightforward, and unlike me you will be able to test it.

2opremio, 2025-03-06:
> this should be straightforward

@mikeller I was planning to look into that, but I don't see a very reliable way based on names.

As you can see in the screenshots above, when using the BT interface with my Cressi Donatello this is what I get:

* The BLE device name is `4_1e52`  (`4_254d` on my gf's Donatello). I am pretty sure it means `<model_number>_<serial_number>` (I have confirmed in the libcomputer code that 4 matches teh Donatello model number).
* The manufacturer name is `B810 SRL` which seems to be https://www.b810group.it/ (likely, the creators of the interface).
* The Device information Service doesn't seem to offer any other meaningful info. In particular, it doesn't provide a PnP ID on which we can rely for a vendor+product. All the other fields also seem to the refer to the interface and not the watch.

_Maybe_ I can match the manufacturer name for `B810 SRL` and the device name for `<number>_<4 hex numbers>`. Thoughts?

A more reliable way would be to check for service UUID `6e400001-b5a3-f393-e0a9-e50e24dc10b8` on discovery and then read the device identification values. But, AFAICT, this seems pretty big deviation from the name-based matching `btdiscovery.cpp` is doin today.

2opremio, 2025-03-06:
Regardless, my priority right now is to get the divelog reading to work.

I am bit lost as to what could be wrong. @jefdriesen any pointers? I will keep digging, though.

mikeller, 2025-03-06:
> All the other fields also seem to the refer to the interface and not the watch.

I think recognising and showing the interface in the filtered list will be good enough - I think most users of this will not have more than one dive computer they want to download from at the same time. So they will naturally tend to select whichever device is showing up in the filtered list.

> _Maybe_ I can match the manufacturer name for `B810 SRL` and the device name for `<model number>_<4 hex numbers>`. Thoughts?

This seems to be an ok approach. Or just matching on one of these two characteristics will be good enough I think.

> Regardless, my priority right now is to get the divelog reading to work.

Agreed - there is not much point in making connecting easy if we can't read the logs. 😆

2opremio, 2025-03-07:
So after further digging, it seems there is a timeout when reading the dive logs.

1. `subsurface` identifies `UUID: Nordic UART Service (6e400001-b5a3-f393-e0a9-e50e24dcca9e)` as the preferred serial service.
2. `libdivecomputer` sends `CMD_LOGBOOK_BLE` (`0x02`) (which `subsurface` sends to the Nordic UART TX characteristic)
3. `libdivecomputer` reads the response (which `subsurface` assumes refers to data previously obtained through notifications from the Nordic UART RX characteristic).
4. `subsurface` waits for data to send down to `libdivecomputer` but nothing arrives within the timeout and returns an error.

(BTW, it would had really helped if `subsurface`/`libdivecomputer` printed actual error codes into the logs).

The only thing I can think of is that `subsurface` is wrongly identifying the Nordic service as the preferred and it should use the two first characteristics of the propietary interface service.

I will give that a try.

2opremio, 2025-03-07:
Yeah, that was it. `subsurface` was picking the Nordic UART service (which is used by the Interface device for something else) instead of the serial service.

As a workaround, I commented out the`Nordic UART` from the preferred service list and I have now managed to successfully import the dives. 

Now I have to find a proper fix

mikeller, 2025-03-07:
> Yeah, that was it. `subsurface` was picking the Nordic UART service (which I guess is used by the Interface device for something else) instead of the actual divewatch serial service.
> 
> As a workaround, I commented out the`"Nordic Semi UART"` from the preferred service list and I have now managed to successfully import the dives.

ruh roh that's a mess. But well done on the find!
What is the  service that is actually working?

> Now I have to find a proper fix for devices with multiple matches in `serial_service_uuids`

If we add the service that works at the top of `serial_service_uuids` in `core/qt-ble.cpp` and treat the list as ordered by preference we should be able to solve this in a somewhat generic way without breaking something else.

@torvalds might know which devices are using the "Nordic Semi UART" as he added this in #2844.

2opremio, 2025-03-07:
> ruh roh that's a mess. But well done on the find! What is the service that is actually working?

```diff
diff --git a/core/qt-ble.cpp b/core/qt-ble.cpp
index f17f26841..8666640cb 100644
--- a/core/qt-ble.cpp
+++ b/core/qt-ble.cpp
@@ -164,6 +164,7 @@ static const struct uuid_match serial_service_uuids[] = {
        { "fdcdeaaa-295d-470e-bf15-04217b7aa0a0", "ScubaPro (G2, G3)"},
        { "fe25c237-0ece-443c-b0aa-e02033e7029d", "Shearwater (Perdix/Teric/Peregrine/Tern)" },
        { "0000fcef-0000-1000-8000-00805f9b34fb", "Divesoft" },
+       { "6e400001-b5a3-f393-e0a9-e50e24dc10b8", "Cressi BT Interface"},
        { NULL, }
 };
```

 

> If we add the service that works at the top of `serial_service_uuids` in `core/qt-ble.cpp` and treat the list as ordered by preference we should be able to solve this in a somewhat generic way without breaking something else.

Unfortunately that's not enough. The result of `BLEObject::addService()` depends on  _whitelisted_ service is discovered (i.e. added ) last, since it clears all the previously added services.

https://github.com/subsurface/subsurface/blob/e637f2476df4e826b787b03bf5951176f895e7a1/core/qt-ble.cpp#L209-L236

In my case, the Nordic service was discovered last, clearing the `Cressi BT Interface` service, which was previously added.

I cannot think of a clean solution without  risking breaking users.

We could maybe add priorities so that `addService()` doesn't clear the list if a service with a higher priority is found.

Or, even simpler, mark some services as unique within a watch, so that they cannot be cleared by `BLEObject::addService()` after they are added.

The ordering impact of `addService()` is pretty hideous.

An even better fix is never clearing _whitelisted_ services and then, after discovery, choosing only one based on it's ordering in `serial_service_uuids`. I think that should remove the ordering side-effects. I am not sure about user impact, though.

> [@torvalds](https://github.com/torvalds) might know which devices are using the "Nordic Semi UART" as he added this in [#2844](https://github.com/subsurface/subsurface/pull/2844).

I am curious 🍿

2opremio, 2025-03-07:
@mikeller OK, I think I have a decent fix (see https://github.com/2opremio/subsurface/commit/f7ae1e9bf50ba22a2ac9c77d0cc2257fef92fe93 ) but can we get #4463 merged first?

mikeller, 2025-03-07:
> [@mikeller](https://github.com/mikeller?rgh-link-date=2025-03-07T04%3A55%3A23.000Z) OK, I think I have a decent fix (see [2opremio@f7ae1e9](https://github.com/2opremio/subsurface/commit/f7ae1e9bf50ba22a2ac9c77d0cc2257fef92fe93) ). However, can we get [#4463](https://github.com/subsurface/subsurface/pull/4463) merged before I create another PR?

Looks good to me - that's pretty much what I had in mind.

No problem mergeing #4463 first. I normally let pull requests stick around for at least 24 hours (unless they are trivial fixes). The people contributing to Subsurface are scattered around the globe, so this ensures that everybody gets the opportunity to provide feedback at a reasonable time within their timezone.

jefdriesen, 2025-03-07:
Looks like you already found the problem while I was sleeping. Too late now, but I maintain a page with all known BLE services and characteristics:

https://github.com/libdivecomputer/libdivecomputer/wiki/BLE

That could have saved you some time.

Anyway, it's strange they have the standard Nordic UART service and then add another vendor specific which basically also provides another virtual uart. Maybe the Nordic service is used for firmware upgrades or communication with the PC interface itself?

2opremio, 2025-03-07:
> Looks like you already found the problem while I was sleeping. Too late now, but I maintain a page with all known BLE services and characteristics:
> 
> https://github.com/libdivecomputer/libdivecomputer/wiki/BLE

It's surprising so many devices use the Nordic UART as is.

> That could have saved you some time.

indeed

> Anyway, it's strange they have the standard Nordic UART service and then add another vendor specific which basically also provides another virtual uart. Maybe the Nordic service is used for firmware upgrades or communication with the PC interface itself?

I am not sure what's exactly for, but I doubt it's for firmware upgrades since it also exposes the Nordic DFU service.

2opremio, 2025-03-07:
I have now pushed https://github.com/subsurface/subsurface/pull/4464 too (the latest simplification to #4463 removed all the conflicts, allowing for independent reviews/merging)

2opremio, 2025-03-07:
I will now work on another PR to support service discovery, as agreed with @mikeller.

mikeller, 2025-03-08:
> I am not sure what's exactly for, but I doubt it's for firmware upgrades since it also exposes the Nordic DFU service.

The DFU service is exactly what is intended for firmware upgrades of the bluetooth adaptor - so I suspect it is intended for this.
In theory, DFU is intended to be used as an inline capability, i.e. the communications channel should remain available for 'normal' communication even if it has DFU capability (which is usually only active for a limited time directly after power-up or reset), but it looks like the designers of this device didn't trust that..

2opremio, 2025-03-10:
I've implemented basic name-based device discovery at https://github.com/subsurface/subsurface/pull/4469

That should be the last PR to provide (at least initial) support for the Cressi Bluetooth interfaces.

closed 2025-03-11 by mikeller