subsurface/subsurface #4462: Macos Verison cannot plan CCR dives

f0xdude, 2025-03-06:
In the latest macos version when the user goes into plan dive mode and selects CCR from the dropdown the program does not allow setting setpoint.
this makes it impossible to use the program on macos to plan ccr diving.

mikeller, 2025-03-06:
The 'latest macos version' is a bit vague. What version of Subsurface and MacOS are you using?

Entering setpoints is working just fine for me here with Subsurface 6.0.5347 on MacOS 12.7.6 (Intel silicone):

[image]

f0xdude, 2025-03-06:
I apologize.
My macos is "15.3.1 (24D70)"  M2 apple hardware.
I tried the v6.0.5299

The setpoint dropdowns are missing or not rendered and not clickable.
Rebreather:bailout / Deco on OC also not affecting anything.
When CCR is selected subsurface still calculates regular OC dive on the right

[image]

[image]

f0xdude, 2025-03-06:
oh wow that is new behabior, older verisons i belive were assuming the gases without their designations. 
In any case, thatnk you please close the issue as it's user error instead of actual bug.

I'm sorry for raising a false alarm.

closed 2025-03-06 by f0xdude