subsurface/subsurface #4465: SAC rate for Garmin Descent calculated on gas volume for all transmitters ...

bwong2132, 2025-03-07:
As you may know, a Garmin Descent MK2i or MK3i can track up to 8 transmitters. In my case, I have a transmitter for my tank and another for my wife's. If I got the calculations right, I believe Subsurface takes the gas consumption from both transmitters --- e.g., 

   ((172.92-13.51) + (199.67-55.98)) x 11.1 
   __________________________________. / 49 (duration).   which is about 28.97.   
                       2.37 (ATA)

The average depth was 13.7m so ATA of 2.37

The SAC for just my tank would be 

(199.67-55.98) x 11.1
___________________      / 49 or   13.7
           2.37

I was wondering -- would it be possible to 

- identify my tank vs. the tanks I watch (e.g., wife and daughter). 
- calculate my SAC rate -- I don't believe you have the other tank's duration or average depth
- be able to display all the tanks -- but high light my tank to distinguish it from the rest. 

Thanks in advance. And thanks for a great dive log app. 

Bernie

bwong2132, 2025-03-07:
I am uploading a FIT file -- just rename .txt to .fit

[2025-03-06-12-10-45.txt](https://github.com/user-attachments/files/19134303/2025-03-06-12-10-45.txt)

mikeller, 2025-03-07:
I am currently working on a way (#4461) that will allow you to 'ignore' the data from your wife's transmitter in your dive log - this should help with the SAC calculation.
At the moment, Subsurface does not contain any functionality to track dive data from a second diver in a dive log, and I don't think there is sufficient demand to add this.
But once #4461 has been completed, you could always create a separate dive log for your wife, and then import the dives from your dive computer that include 'her' transmitter, and then delete 'your' transmitter in this log.

bwong2132, 2025-03-09:
Thanks Mike - looking forward to that change. I would be happy to beta test
it.
Bernie

mikeller, 2026-10-17:
Hi @bwong2132,

The core problem you raised, SAC being inflated because a second (your wife's) transmitter's data was being included, has been addressed: PR #4461 "Desktop: Improve Management of Tank Sensors" added a way to exclude an unwanted sensor's data from a dive's gas-consumption accounting.

On the broader idea of tracking a second diver's full dive data separately: as explained in the thread, we don't think there's enough demand to justify building that right now. The workaround (a separate dive log for the other diver, importing the same dive computer data and removing the irrelevant transmitter per-log) still applies if you want it.

Closing this as the core SAC issue is fixed. Please reopen if the tank-sensor management in PR #4461 doesn't give you a way to exclude the unwanted transmitter's data on a current build.

closed 2026-10-17 by mikeller