subsurface/subsurface #4475: Mares Sirius and 6.0.5357

skaggerik, 2025-03-19:
Windows 11 24H2: 
Sirius does not appear in Subsurface BT search list, not even with "Show all BT devices" checked. 
Sirius does appear in Windows bluetooth connection box. When attempting to connect, sirius shows connected only for a few seconds. Windows does not seem to be able to pair successfully. 
Bluetooth adapter: Mediatek (HP laptop)

Issue on Android: 
"No new dives to download" (translated from Swedish)
"Force download all dives" (translated from Swedish) does not change outcome. 
Google pixel 8
Android 15

Sirius has no updates to install according to Mares app.

skaggerik, 2025-03-19:
Diving Log will connect on both PC and phone.

rmultan, 2025-03-20:
Hi, it works on MacOs 15.1.1

[image]

It should work since https://github.com/subsurface/subsurface/pull/4384 has been merged

rmultan, 2025-03-21:
It appears that the download doesn't work anymore

skaggerik, 2025-03-23:
Tried with 6.0.5358. No change on android. TV appeared in list on Windows, no other changes.

skaggerik, 2025-03-30:
Tried updating PC and phone to 6.0.5365, no change. I did however notice it saying "Sirius firmware=0". My sirius is 01.06.16
Windows application detects every BT device except for sirius. 

[image]

njaehn, 2025-04-07:
Hi, I've got the same issue. I found this Google-Groupe entry but. I didn't know what to do with it.
https://groups.google.com/g/subsurface-divelog/c/MWRdzl6jqRk

skaggerik, 2025-04-07:
v6.0.5371: Still broken

skaggerik, 2025-04-17:
6.0.5378: Works on android

njaehn, 2025-04-30:
Ok I've tried many Version including the 6.0.5378 and on my S23 Ultra I get a connection but it still can't download any dives.

mankshank, 2025-05-01:
I also am having issues, both on desktop (linux mint) as well as android (graphene os on a pixel 8). I am on the latest firmware (01.06.16), and have updated to enable freedive mode. These problems existed before I changed that though. It seems like both devices see the computer fine, and there is no issue with bluetooth or the connection.

On desktop version 6.0.5387-CICD (and the two previous) I get the message "Error parsing the header: Data format error" for all 3 of the dives on the dive computer, and nothing is imported. 

On mobile it just says "no new dives downloaded from dive computer".

Attached is the log from desktop. 

[subsurface.log](https://github.com/user-attachments/files/20004117/subsurface.log)

jefdriesen, 2025-05-20:
> Windows 11 24H2: Sirius does not appear in Subsurface BT search list, not even with "Show all BT devices" checked. Sirius does appear in Windows bluetooth connection box. When attempting to connect, [...]

On Windows, subsurface can only detect bluetooth devices that have been paired using the standard windows tools. Devices which are not paired can't be enumerated. This is a limitation of the bluetooth api used by subsurface.

> On desktop version 6.0.5387-CICD (and the two previous) I get the message "Error parsing the header: Data format error" for all 3 of the dives on the dive computer, and nothing is imported.

This error is already fixed in [libdivecomputer](https://github.com/libdivecomputer/libdivecomputer/commit/45fcb6826a7579be15b8c08661f6bee7e919ab7d), but the fix is not yet included in the subsurface builds.

mankshank, 2025-05-20:
> This error is already fixed in [libdivecomputer](https://github.com/libdivecomputer/libdivecomputer/commit/45fcb6826a7579be15b8c08661f6bee7e919ab7d), but the fix is not yet included in the subsurface builds.

thanks for the update. in that case, we should expect this to be implemented in one of the next builds?

skaggerik, 2025-05-23:
> On Windows, subsurface can only detect bluetooth devices that have been paired using the standard windows tools. Devices which are not paired can't be enumerated. This is a limitation of the [...]

My sirius is on it's way to Italy for repairs, but I'm getting the same problem on a Mares Quad Ci. Do you have a workaround for pairing the computer to windows?

skaggerik, 2025-05-23:
> On Windows, subsurface can only detect bluetooth devices that have been paired using the standard windows tools. Devices which are not paired can't be enumerated. This is a limitation of the [...]

Diving log seems to consider the computer being paired. 

[image]

jefdriesen, 2025-05-23:
If it's paired, then subsurface should be able to detect it. 

Note that Diving Log uses a different bluetooth api which can also detect and use bluetooth devices without pairing.

skaggerik, 2025-05-24:
> If it's paired, then subsurface should be able to detect it.
> 
> Note that Diving Log uses a different bluetooth api which can also detect and use bluetooth devices without pairing.

Any advice on what to do?

mankshank, 2025-05-26:
> This error is already fixed in [libdivecomputer](https://github.com/libdivecomputer/libdivecomputer/commit/45fcb6826a7579be15b8c08661f6bee7e919ab7d), but the fix is not yet included in the subsurface builds.

thanks for the info! 

does this mean that it is only a matter of time for it to be included in the next release, or is there more testing/work that needs to be done before it gets implemented?

skaggerik, 2025-05-26:
> Do you see the device in the Windows control panel as a bluetooth device? If it's not listed there, then it's not paired and won't be detected by subsurface. If it's shown, can you share a subsurface download log? The bluetooth device discovery info might give some clues.
> 
> I'm not a Windows user (except in VM's for testing libdivecomputer), and don't have any Mares dive computers, so I can't give much advice.

It won't pair properly, I've discovered. However, subsurface can discover bluetooth devices other than Sirius and the quad ci (if I check Show all bt devices). Exempel: a Tvs, a phone etc. It feels almost like there is a filter for my mares computers.

skaggerik, 2025-05-27:
> > does this mean that it is only a matter of time for it to be included in the next release, or is there more testing/work that needs to be done before it gets implemented?
> 
> Usually the changes are merged into the subsurface fork from time to time.
> 
> > It won't pair properly, I've discovered. However, subsurface can discover bluetooth devices other than Sirius and the quad ci (if I check Show all bt devices). Exempel: a Tvs, a phone etc. It feels almost like there is a filter for my mares computers.
> 
> If the "Show all bt devices" isn't checked, then subsurface will indeed hide bluetooth devices that don't match the selected dive computer. If it is checked, then all devices that have been discovered are shown.

I'm sorry, but I fail to understand why subsurface can detect several non-paired devices, but not my dive computers.

jefdriesen, 2025-05-27:
Can you maybe share a screenshot of the bluetooth device list in Windows and Subsurface, along with the log file from subsurface.

skaggerik, 2025-05-27:
> Can you maybe share a screenshot of the bluetooth device list in Windows and Subsurface, along with the log file from subsurface.

[image]
[image]

How do I generate log files from subsurface?

jefdriesen, 2025-05-27:
The Windows screenshot shows the "Add bluetooth device" dialog. This shows the bluetooth devices that have been found during the discovery. You still need to select the device here, to add it to the system. Once that is done, it should also appear in subsurface.

jefdriesen, 2025-05-27:
The link to the exact commit is in my comment (which you quoted). The confusion is probably because that change is very similar to the fix you already cherry-picked, but it's another change. There is a version in the profile, but also in the header.

jefdriesen, 2025-05-27:
> Problem: it wont pair properly with windows. Windows shows "connecting", and then just gives up. Same issue on another PC.

Unfortunately I don't know why that fails or how to fix it. Sorry.

mikeller, 2025-05-27:
> The link to the exact commit is in my comment (which you quoted). The confusion is probably because that change is very similar to the fix you already cherry-picked, but it's another change. There is a version in the profile, but also in the header.

https://stackoverflow.com/questions/35007939/what-is-the-difference-between-commits-and-pull-requests

jefdriesen, 2025-06-19:
The changes have been merged, so try the latest subsurface build.

mikeller, 2026-10-17:
Hi @skaggerik,

Two separate things came up in this thread, so to wrap up both:

1. The header-parsing failure after your Sirius firmware update to 1.6.16 (header v2.0): this is fixed. I've checked libdivecomputer/src/mares_iconhd_parser.c directly in current source and the header-version check jefdriesen described is in place.
2. The Windows BLE pairing issue (device not appearing even with "show all BT devices"): as jefdriesen explained, this is a platform limitation, not a Subsurface bug, on Windows, Subsurface can only detect Bluetooth devices already paired via Windows' own Bluetooth settings. If Windows itself won't complete pairing with your device, that's a Windows/driver issue outside what we can fix directly.

Closing this as fixed for the parsing bug. Please reopen if you still see a header-parsing error for Sirius dives on a current build.

closed 2026-10-17 by mikeller