subsurface/subsurface #4481: SAC not calculated for XR dives with two or more deco cyclinders

rsettembrini, 2025-03-24:


atdotde, 2025-03-25:
You need to give us a bit more information. What is a "XR dive"? In order for the SAC to be be calculated, Subsurface needs the initial and final pressures of all cylinders used as well as their volumes. If anything is missing, it cannot figure out the SAC. Also for rebreather dives it does not make sense to compute a SAC. Can you share a dive where this happens (use the Export function from the file menu)?

Best
Robert

mikeller, 2026-10-17:
Hi @rsettembrini,

Thanks for reporting the SAC calculation issue with multiple deco cylinders. To track down the specific trigger, we'd need the exact cylinder-switch pattern involved, or ideally a sample dive (exported file) that reproduces the wrong SAC value, there are several ways a dive can end up with 2+ deco cylinders and we need to see which one is causing this.

If we don't get a sample within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.