subsurface/subsurface #4489: logtrak

cakepans, 2025-03-28:
I have an exported a logtrak file from the scubapro logtrak2.0 app but when I try to import the dives the tags I am able to add do not match up with the json names of the stats from my dive. 

Here is an example group from the raw file. 

"label":null,"utcDifferenceMinutes":-225,"rating":null,"buddyNames":[],"guideNames":[],"notes":null,"weight":null,"conditionWeather":null,"conditionVisibility":null,"conditionWaves":null,"mainImageId":null,"showMainImageInLogbook":false,"images":[],"equipmentIds":[],"startTime":"2023-09-04T20:13:35.000Z","endTime":"2023-09-04T21:15:35.000Z","diveComputerId":"c3AzlUkOvPQ70GkuRgMa","saltwaterCalibrated":true,"locationName":null,"locationSpot":null,"locationLat":null,"locationLng":null,"depthMetersMax":12.64,"waterTempCelsiusMin":29.6,"waterTempCelsiusMax":34.4,"diveMode":"scuba","diveSettings1":269533184,"diveSettings2":16777215,"diveSettings3":null,"decoInformation":null,"tankData":{"tank1":{"o2Mixture":30,"heMixture":null,"startPressure":null,"endPressure":null,"maxPp02Limit":1.4,"minPp02Limit":null,"pairingAddress":null,"size":null,"type":null,"aggregatedAmbientPressure":1964146},"tank2":null,"tank3":null,"tank4":null,"tank5":null,"tank6":null,"tank7":null,"tank8":null,"tank9":null,"tank10":null},"lastSendToSsi":null}

is it possible to get around this other than doing a manual log?

mikeller, 2025-03-29:
@cakepans: What version of Subsurface are you using?

cakepans, 2025-03-29:
current stable windows version as of date of ticket creation.

mikeller, 2025-03-29:
@cakepans: Are you able to be a bit more specific, or are you expecting everybody who reads this and could potentially help to put in the effort of trawling through the release history?

cakepans, 2025-03-29:
V6.0.5214

mikeller, 2025-03-29:
Try upgrading to the latest nightly build from https://subsurface-divelog.org/latest-release - there was a bugfix that might have resolved this (#4479).

cakepans, 2025-03-29:
Tried out the latest nightly build
(subsurface-6.0.5367-CICD-release-installer.exe)
no luck the tags still don't match up. it seems the file type (.logtrak) is
not accepted.

cakepans, 2025-04-15:
> @cakepans <https://github.com/cakepans>: Have a look at the section of
> the user manual dealing with importing from LogTrak, and try to use one of
> the methods described there to export from LogTrak / import into Subsurface.

It appears the ".logtrak" filetype is not supported.
The logtrak log files that end in ".ASD" or ".script" are supported by
Subsurface. Those files come from logtrak windows/mac programs .
I got this ".logtrak" file from the Scubapro app called, Logtrak2.0, via
its database backup function.
I did try to import the file in the way you suggested but it's not
compatible.

mikeller, 2026-10-18:
Hi @cakepans,

This would need a new parser path distinct from the existing LogTrak ASD/script support, since the Logtrak 2.0 app's JSON export is a genuinely different format. It's not urgent given this is currently the only report we've had of it, but your JSON sample is a solid starting reference if anyone picks it up. Let us know if you're still hitting this, or if more Logtrak 2.0 users turn up with the same gap, and we'll prioritise it accordingly.