subsurface/subsurface #4490: Merging dives doesn't marge tanks

anubisg1, 2025-04-01:
When merging two dives, the tanks are not merged.

the example dives are here:

```
<divelog program='subsurface' version='3'>
<settings>
<fingerprint model='1343290d' serial='00005227' deviceid='308c9f95' diveid='7c19bcff' data='15002c00190002007d00'/>
<fingerprint model='552a3cae' serial='0000a119' deviceid='2f1ee3a8' diveid='4a2ee4cd' data='0b0034001c0006007b00'/>
<fingerprint model='6b6ca4de' serial='ab1fa4c3' deviceid='98e7ba48' diveid='8a20c2e9' data='67e7eb78'/>
</settings>
<divesites>
<site uuid='de47f54b' name='Gulaska - Senec' gps='48.207944 17.425472' description='Lake'>
  <notes>Attractions underwater </notes>
  <geo cat='2' origin='0' value='Slovakia'/>
  <geo cat='3' origin='0' value='Bratislavský Kraj'/>
  <geo cat='5' origin='0' value='Krmeš'/>
  <geo cat='6' origin='3' value='Krmeš'/>
</site>
</divesites>
<dives>
<dive number='55' visibility='3' current='5' sac='19.913 l/min' divesiteid='de47f54b' date='2025-03-29' time='11:51:01' duration='48:00 min'>
  <suit>Dry suit 3 lam.</suit>
  <cylinder size='11.094 l' workpressure='206.843 bar' description='AL80' end='152.236 bar' />
  <cylinder size='11.094 l' workpressure='206.843 bar' description='AL80' end='130.725 bar' />
  <weightsystem weight='10.0 kg' description='weight' />
  <divetemperature air='12.0 C'/>
  <divecomputer model='Shearwater Perdix 2' deviceid='98e7ba48' diveid='bb207f5f'>
  <depth max='3.8 m' mean='3.03 m' />
  <temperature water='7.0 C' />
  <surface pressure='1.002 bar' />
  <water salinity='1020 g/l' />
  <extradata key='Logversion' value='16(PNF)' />
  <extradata key='Serial' value='ab1fa4c3' />
  <extradata key='FW Version' value='99' />
  <extradata key='Deco model' value='GF 30/70' />
  <extradata key='Battery type' value='1.5V Lithium' />
  <extradata key='Battery at end' value='1.5 V' />
  <extradata key='T1 battery' value='normal' />
  <extradata key='T2 battery' value='normal' />
  <event time='0:10 min' type='25' flags='1' name='gaschange' cylinder='0' />
  <sample time='0:10 min' depth='1.3 m' temp='9.0 C' pressure0='203.671 bar' pressure1='204.774 bar' tts='1:00 min' />
  <sample time='0:20 min' depth='1.8 m' pressure0='203.395 bar' pressure1='204.636 bar' ndl='99:00 min' />
  <sample time='0:30 min' depth='3.0 m' pressure0='202.43 bar' pressure1='204.499 bar' />
  <sample time='0:40 min' depth='2.8 m' pressure0='201.189 bar' pressure1='204.636 bar' />
  <sample time='0:50 min' depth='2.6 m' pressure0='200.362 bar' />
  <sample time='1:00 min' depth='2.9 m' pressure0='199.81 bar' />
  <sample time='1:10 min' depth='2.7 m' pressure0='199.258 bar' />
  <sample time='1:20 min' depth='2.9 m' pressure0='198.845 bar' pressure1='204.499 bar' />
  <sample time='1:30 min' depth='2.5 m' temp='8.0 C' pressure0='198.293 bar' />
  <sample time='1:40 min' depth='2.6 m' pressure0='197.604 bar' />
  <sample time='1:50 min' depth='2.7 m' pressure0='196.776 bar' />
  <sample time='2:00 min' depth='3.2 m' pressure0='196.363 bar' />
  <sample time='2:10 min' depth='3.2 m' pressure0='195.397 bar' />
  <sample time='2:20 min' depth='2.6 m' pressure0='194.57 bar' />
  <sample time='2:30 min' depth='1.4 m' pressure0='194.294 bar' />
  <sample time='2:40 min' depth='0.0 m' pressure0='194.018 bar' pressure1='204.361 bar' />
  <sample time='2:50 min' depth='1.9 m' pressure0='193.467 bar' />
  <sample time='3:00 min' depth='3.1 m' pressure0='192.777 bar' />
  <sample time='3:10 min' depth='3.5 m' pressure0='192.088 bar' pressure1='204.223 bar' />
  <sample time='3:20 min' depth='3.5 m' pressure0='191.398 bar' />
  <sample time='3:30 min' depth='3.5 m' pressure0='190.847 bar' pressure1='204.361 bar' />
  <sample time='3:40 min' depth='3.4 m' pressure0='190.433 bar' pressure1='204.223 bar' />
  <sample time='3:50 min' depth='3.5 m' pressure0='189.468 bar' />
  <sample time='4:00 min' depth='3.5 m' pressure0='189.606 bar' />
  <sample time='4:10 min' depth='3.3 m' pressure0='188.916 bar' />
  <sample time='4:20 min' depth='3.6 m' pressure0='188.365 bar' pressure1='204.361 bar' />
  <sample time='4:30 min' depth='3.4 m' pressure0='187.4 bar' pressure1='204.223 bar' />
  <sample time='4:40 min' depth='3.3 m' />
  <sample time='4:50 min' depth='3.3 m' pressure0='186.986 bar' />
[... 435 lines cut ...]
  <sample time='24:00 min' depth='1.8 m' pressure1='93.217 bar' rbt='21:00 min' />
  <sample time='24:10 min' depth='1.8 m' pressure0='123.278 bar' pressure1='92.528 bar' rbt='20:00 min' />
  <sample time='24:20 min' depth='1.8 m' pressure0='123.14 bar' pressure1='91.838 bar' rbt='19:00 min' />
  <sample time='24:30 min' depth='1.8 m' pressure1='91.011 bar' rbt='18:00 min' />
  <sample time='24:40 min' depth='1.7 m' pressure1='90.459 bar' rbt='16:00 min' />
  <sample time='24:50 min' depth='1.9 m' pressure1='90.046 bar' rbt='15:00 min' />
  <sample time='25:00 min' depth='1.5 m' pressure1='89.632 bar' rbt='14:00 min' />
  <sample time='25:10 min' depth='0.0 m' pressure1='89.218 bar' rbt='13:00 min' />
  <sample time='25:20 min' depth='1.1 m' pressure1='88.667 bar' />
  <sample time='25:30 min' depth='1.4 m' pressure0='123.278 bar' pressure1='88.253 bar' />
  <sample time='25:40 min' depth='1.5 m' pressure0='123.14 bar' pressure1='87.839 bar' />
  <sample time='25:50 min' depth='1.6 m' pressure0='123.278 bar' pressure1='87.288 bar' rbt='12:00 min' />
  <sample time='26:00 min' depth='1.9 m' pressure0='123.14 bar' pressure1='86.736 bar' />
  <sample time='26:10 min' depth='1.1 m' pressure1='86.598 bar' rbt='11:00 min' />
  <sample time='26:20 min' depth='0.0 m' pressure1='86.322 bar' rbt='12:00 min' />
  <sample time='26:30 min' depth='0.0 m' pressure0='123.278 bar' pressure1='86.047 bar' tts='0:00 min' rbt='13:00 min' />
  <sample time='26:40 min' depth='0.0 m' pressure0='123.14 bar' pressure1='85.495 bar' />
  <sample time='26:50 min' depth='0.0 m' pressure1='85.357 bar' rbt='14:00 min' />
  <sample time='27:00 min' depth='0.0 m' pressure0='123.002 bar' pressure1='85.633 bar' rbt='16:00 min' />
  <sample time='27:10 min' depth='0.0 m' pressure0='121.761 bar' pressure1='85.771 bar' rbt='17:00 min' />
  </divecomputer>
</dive>
</dives>
</divelog>
```

mikeller, 2025-04-15:
@anubisg1: Can you please check with one of the artifacts in #4504 if this works in the way you expect it?

anubisg1, 2025-04-16:
hello @mikeller 

i installed this version 

[image]

i imported the dive log (same xml as above) and i get the 2 dives:

[image]

here are the tanks, dive 1

[image]

dive 2 

[image]

when i merge the dives the dive profiles are merged but the problem remains the same...

[image]

notice that tank 1 and 3 are the same tanks... same as tank 2 and 4 (you can see from the pressure end/start 

i would expect the tank pressure and AI data to be merged , but that's not the case

mikeller, 2025-04-16:
> i would expect the tank pressure and AI data to be merged , but that's not the case

No, that is not what Subsurface currently does - all that we are trying to do when dives are merged is to keep all the tank sensors that are assigned showing in the profile.
Tanks are only merged if they have identical start / end pressures, like when overlapping logs from multiple dive computers worn on the same dive are merged.
When logs are merged but their dive times and the tank pressures differ we do not have the logic to make guesses as to which tanks should be merged, so we keep treating them as separate tanks. But at least now all of the tank pressure graphs should be showing in the profile after the merge.

anubisg1, 2025-04-16:
If this case you can see that end pressure of tank 1 is identical to tank pressure of thank 3, same for 2 and 4.

What is stopping us from merging those? If you can't make "automatic" guesses, could the user instead manually decide what tanks to merge?

The first dive in the xml provided was 2 dives merged successfully with the tank as well merged successfully (around minute 23), so a user can't tell what to expect.

anubisg1, 2025-04-16:
I don't have the knowledge necessary to do it, but the question was an honest one. What checks are failing in the current implementation that stops the tanks from merging?

The start/end pressure it's not it

PR #4504 (2025-04-17): Desktop: Fix Tank Sensors after Merging Dives

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fix the assignment of tank sensors after multiple dives have been merged
into one.
This was done a bit too simplistic in #4461.

### Changes made:

### Related issues:

Fixes: #4490

### Additional information:

### Release note:

### Documentation change: