subsurface/subsurface #4495: unable to import dive containing "&" in some field

maurofurini, 2025-04-05:
I exported my dive to divelogs.de.
I'm trying to import in subsurface from divelogs.de.
Im my last cruise the boat name was "Maxene & Danella".
The import crash with "unable to parse" when reach the first dive with boat "Maxene & Danella" (<BOATNAME><![CDATA[Maxene & Danella]]></BOATNAME>)
Importing the XML for this dive after removing the "&" was ok.

mikeller, 2025-04-05:
> The import crash with "unable to parse" when reach the first dive with boat "Maxene & Danella" ()

What was the exact string that appeared in the XML generated by divelogs.de for the boat name?

maurofurini, 2025-04-07:
<BOATNAME>
<![CDATA[ Maxene & Danella ]]>
</BOATNAME>

maurofurini, 2025-04-07:
```html
<?xml version="1.0" encoding="UTF-8"?>
<DIVELOGSDATA>
  <DIVELOGSDIVENUMBER>1389</DIVELOGSDIVENUMBER>
  <DIVELOGSID>4019011</DIVELOGSID>
  <COMPUTERMODEL>iX3M 2 Tech+</COMPUTERMODEL>
  <DATE>14.02.2025</DATE>
  <TIME>12:12:00</TIME>
  <DIVETIMESEC>3180</DIVETIMESEC>
  <SURFACETIME>4500</SURFACETIME>
  <MAXDEPTH>22.3</MAXDEPTH>
  <MEANDEPTH>14.8</MEANDEPTH>
  <LOCATION><![CDATA[Filippine, Malapasqua]]></LOCATION>
  <SITE><![CDATA[Gato Island - West Wall]]></SITE>
  <WEATHER><![CDATA[Sereno]]></WEATHER>
  <WATERVIZIBILITY><![CDATA[Media]]></WATERVIZIBILITY>
  <AIRTEMP>28.5</AIRTEMP>
  <WATERTEMPMAXDEPTH>27.2</WATERTEMPMAXDEPTH>
  <WATERTEMPATEND>0.0</WATERTEMPATEND>
  <PARTNER><![CDATA[Laura Baraldini]]></PARTNER>
  <BOATNAME><![CDATA[Maxene & Danella]]></BOATNAME>
  <CYLINDERNAME><![CDATA[]]></CYLINDERNAME>
  <CYLINDERDESCRIPTION><![CDATA[Alluminio]]></CYLINDERDESCRIPTION>
  <DBLTANK>0</DBLTANK>
  <CYLINDERSIZE>12.00</CYLINDERSIZE>
  <CYLINDERSTARTPRESSURE>194.00</CYLINDERSTARTPRESSURE>
  <CYLINDERENDPRESSURE>75.00</CYLINDERENDPRESSURE>
  <WORKINGPRESSURE/>
  <WEIGHT>8.00</WEIGHT>
  <O2PCT>21.0</O2PCT>
  <HEPCT/>
  <LOGNOTES><![CDATA[]]></LOGNOTES>
  <LAT>11.446361</LAT>
  <LNG>124.019196</LNG>
  <GOOGLEMAPSZOOMLEVEL>14</GOOGLEMAPSZOOMLEVEL>
  <SAMPLEINTERVAL>10</SAMPLEINTERVAL>
  <SAMPLE>
    <DEPTH>5.9</DEPTH>
    <TEMP>28.5</TEMP>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>7.3</DEPTH>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>8</DEPTH>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>9.7</DEPTH>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>11.4</DEPTH>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>12.1</DEPTH>
    <TEMP>28.3</TEMP>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>12.6</DEPTH>
    <TEMP>28.4</TEMP>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>13</DEPTH>
    <TEMP>28.3</TEMP>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>13.9</DEPTH>
    <TEMP>28.2</TEMP>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>14.7</DEPTH>
[... 936 lines cut ...]
  </SAMPLE>
  <SAMPLE>
    <DEPTH>3.7</DEPTH>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>3.3</DEPTH>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>3.2</DEPTH>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>2.4</DEPTH>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>1.4</DEPTH>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>0.6</DEPTH>
  </SAMPLE>
  <SAMPLE>
    <DEPTH>0</DEPTH>
  </SAMPLE>
</DIVELOGSDATA>

```

maurofurini, 2025-04-07:
that's the full contents of 1389.xml file