subsurface/subsurface #4496: Can't delete transmitter data from first transmitter

RalfMengwasser, 2025-04-06:
The first transmitter cant be deleted, it has a red line through it. The others can be deleted.

Thats not good, because the first transmitter actually is the one of my girlfriend that has nothing to do with my own gas consumption.

So either just remove the "cant delete" or allow resorting of the transmitters or even let me unselect it on import. Or let me click an "ignore" checkbox?

I dont understand why it makes sense to disallow deletion of one transmitter, really. What is the point behind that? Can I circumvent it in any other way?

mikeller, 2025-04-06:
@RalfMengwasser:

Can you please provide the following:
- version of Subsurface the problem is happening with
- operating system the problem is happening with
- operating system version the problem is happening with
- sequence of steps to reproduce the problem

mikeller, 2026-10-17:
Hi @RalfMengwasser,

Thanks for reporting that you can't delete the first transmitter's data. Could you give us the exact repro steps, which dive(s), how many transmitters/cylinders are involved, and what you click/try when attempting the delete? That'll help us reproduce it on our end.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please comment or reopen whenever suits.