subsurface/subsurface #4497: Combobox selection of location does not work if textbox is empty

RalfMengwasser, 2025-04-06:
If you currently have no selection in the location box and then select a dropdown item, nothing happens.

In order to make it work, the only workaround is to first write random chars into the textbox, accept that and THEN choose the dropdown box item, which then overwrites the random chars.

Thats on a Google Tablet.

mikeller, 2025-04-06:
@RalfMengwasser:

Can you please provide the following:
- version of Subsurface the problem is happening with
- operating system the problem is happening with
- operating system version the problem is happening with
- sequence of steps to reproduce the problem

mikeller, 2026-10-17:
Hi @RalfMengwasser,

Thanks for reporting the combobox location-selection issue. To reproduce this we'd need the exact repro steps, which dialog/field this is in, and ideally a short screen recording or step-by-step sequence showing the textbox being empty and the combobox selection failing.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.