subsurface/subsurface #4513: [Bug]: discontinuity in cylinder pressure with multiple cylinders

livnev, 2025-05-05:
### Description

I'm not sure if this is a bug or if I'm doing something wrong:

[image]

I'm in OC mode and have planned a dive with two gasses, and then imported the actual dive and merged them. The dive computer used was Suunto D4i in freedive mode, meaning it just tracks the dive profile but has no pressure data or deco. I then entered the actual ending pressures and manually added the gas switch event at the right time.

As you can see above the first cylinder now has a weird discontinuity at exactly the run time that my plan had a gas change, jumping by 24 bar, and then quickly going down again before the added gas change. Clearly it's something to do with the interaction between the cylinders and the plan, but unclear if it's a bug or if I have used the software incorrectly

If I remove the gas change I added, then I get this:

[image]

which has that jump at the planned gas change time, no actual gas change, and an incorrect (?) end pressure for the back gas.

### Subsurface Version

6.0.5231

### Platform(s)

Linux

### Platform Version

nixOS 24.11

### Divecomputer Model

Suunto D4i

### Steps to Reproduce

1. Open subsurface
2. Plan dive with two gasses, back gas and deco gas. Subsurface schedules a gas switch.
3. Import dive profile from computer.
4. Edit time of planned dive to agree with imported dive.
5. Select both the plan and the dive and select "Merge dives".
6. Add ending pressures for both cylinders.
7. Add a gas switch event by clicking on the point in the profile where the switch occurred, based on run time.  

### Logs

```shell

```

livnev, 2025-05-05:
@mikeller thanks for your reply. I tried with a more recent version and looks like it's fixed

[image]

As you can see it behaves in a more reasonable way now.

By the way, I assumed what I was doing is supported. I'm just using the freediving mode of one of my computers as a depth gauge and importing it to combine with plan and manually supplied gas info. Is that not supported?

closed 2025-05-05 by livnev

mikeller, 2025-05-05:
> By the way, I assumed what I was doing is supported. I'm just using the freediving mode of one of my computers as a depth gauge and importing it to combine with plan and manually supplied gas info. Is that not supported?

Well, first of all it seems to work for you. 😉
But I suspect it is not a way of using that developers will be focusing on when testing changes to gas management, or to freediving support. So it may or may not be broken by future changes.
Out of interest, why don't you just import the log from the dive computer that is tracking your decompression obligation as well - Subsurface can combine logs from multiple dive computers into a single dive, and this way you'll get accurate gas switch information, and the decompression obligation that is actually calculated and shown in real time during the dive.