subsurface/subsurface #4517: [Feature Request]: Picture support on Dive Site

ekelly, 2025-05-11:
### Problem Statement

I need help remembering the underwater map of the dive.

### Description

It would be great if I could associate a dive site with pictures. Specifically, I want to be able to add a map to the dive site description, so that I can easily refer to it and remember the underwater layout of the dive. This could also be useful for pictures of the surface of the dive site (parking, bathrooms, etc.) which may be useful when returning to the dive site.

I know that I can attach pictures to individual dives, but I would prefer to associate the pictures to the dive site instead so I don't have to remember which dive had the picture of the dive site map in it.

### Alternative Solutions

Attach the map to the first dive at the dive site.

mikeller, 2026-10-18:
Hi @ekelly,

This is a genuine data-model extension (dive sites currently have no picture field at all, unlike dives), so before committing to the moderate scope involved, are you still interested in this, and is it something you (or others) would use regularly? Confirming continued interest will help us prioritise it against other requests.

Happy to pick this up if there's confirmed appetite for it.

ekelly, 2026-10-18:
I am absolutely still interested in this and would use it, primarily with my local dive sites. The lack of this feature is not something that would stop me from using subsurface, but it is a quality of life improvement.