subsurface/subsurface #4525: [Bug]: Calc exceeds time when using 100%, but not 99%

harrydevil, 2025-05-21:
### Description

When planning a dive, using 100% gives an error: "Decompression calculation aborted due to excessive time," but the same dive can be planned with 80% or even 99%.

Screenshots are attached, but I've experimented with the tanks, depths, and run times, and I've experienced the same issue. This issue happens as long as my bottom final depth is above 300. 

[image]
[image]

### Subsurface Version

6.0.5365-CICD-release

### Platform(s)

macOS

### Platform Version

OS X Sequoia 15.4.1

### Steps to Reproduce

1) Plan a dive
2) Add mixes: D 120 HP with 10/60, AL80 15/55, AL80 21/35, AL80 50/0, AL40 100/0
3) Add depths: 150 for 3, 400 for 4, 400 for 2 

### Logs

```shell

```

atdotde, 2025-05-22:
Cannot reproduce this with 6.0.5365.

[image]

harrydevil, 2025-05-22:
Very weird, I was able to recreate it in a new session, please see the attachment.  Also, it is not related to 100%, as the error popped up when I added any additional tank.  I was experimenting and adding a tank of 50/50 allowed it to recalculate, although 21/35, 15/50, and 50/0 (standard gases) gave an error. Sadly, verbose mode doesn't provide additional information.  Is there a log hidden somewhere that might shed more light on what part of the equation causes the error?

*Please note, I was just running this for fun and thought it was helpful to submit the bug, I don't imagine this is a big user issue.

[image]
[image]

[image]

PR #4527 (2025-05-25): Ignore infinite deco in backgas breaks
When doing backgas breaks we might breath a gas that does not advance the decompression. Still a very long or infinite stop time should not abort the deco as we will be switching back to O2.

Fixes #4525

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

### Additional information:

### Release note:

### Doc
commit 22a92e20 (2025-05-25): Ignore infinite deco in backgas breaks

When doing backgas breaks we might breath a gas that does not advace
the deompression. Still a very long or infintite stop time should
not abort the deco as we will be switching back to O2.

Fixes #4525
closed 2025-05-25 by atdotde