subsurface/subsurface #4529: [Feature Request]: Import Tusa DC Solar logs

0x0ACB, 2025-05-27:
### Problem Statement

The Tusa DC Solar is currently not supported by Subsurface. 

### Description

While connecting to the computer via Blutooth might be difficult. It is possible to export the dive logs from the computer which gives you a zip file that contains an sqlite DB with all the dive data

[jp-tabata-divinglog-backup.zip](https://github.com/user-attachments/files/20455117/jp-tabata-divinglog-backup.zip)

### Alternative Solutions

I have created a basic python script to conver the sqlite db into subsurface xml:

```python
#!/usr/bin/env python3
"""
Tusa sqlite to Subsurface XML Converter
Converts diving log data from Tusa sqlite database to Subsurface XML format.
"""

import sqlite3
import xml.etree.ElementTree as ET
from datetime import datetime
import argparse
import sys
from pathlib import Path

class DivingLogConverter:
    """Converter class for transforming Tusa sqlite diving logs to Subsurface XML format."""

    def __init__(self):
        self.root = None
        self.divelog = None
        self.dives = None

    def create_xml_structure(self):
        """Create the basic XML structure for Subsurface format."""
        self.root = ET.Element("divelog")
        self.root.set("program", "subsurface")
        self.root.set("version", "3")

        settings = ET.SubElement(self.root, "settings")

        self.dives = ET.SubElement(self.root, "dives")

    def parse_timestamp(self, timestamp_str):
        """Parse timestamp from milliseconds to datetime format."""
        try:
            if timestamp_str is not None:
                if isinstance(timestamp_str, (int, float)):
                    timestamp_ms = timestamp_str
                else:
                    timestamp_str = str(timestamp_str).strip()
                    if not timestamp_str:
                        return None
                    timestamp_ms = float(timestamp_str)

                timestamp_s = timestamp_ms / 1000
                dt = datetime.fromtimestamp(timestamp_s)
                return dt.strftime("%Y-%m-%d %H:%M:%S")
        except (ValueError, TypeError):
            pass
        return None

    def parse_profile_data(self, depth_str, temp_str):
        """Parse depth and temperature profile data."""
        depth_points = []
        temp_points = []

[... 321 lines cut ...]
    parser = argparse.ArgumentParser(
        description="Convert diving log SQLite database to Subsurface XML format"
    )
    parser.add_argument("db_file", help="Path to the SQLite database file to convert")
    parser.add_argument("-o", "--output", help="Output XML file path (optional)")

    args = parser.parse_args()

    if not Path(args.db_file).exists():
        print(f"Error: File '{args.db_file}' does not exist")
        sys.exit(1)

    converter = DivingLogConverter()
    success = converter.convert_db_to_xml(args.db_file, args.output)

    if success:
        print("Conversion completed successfully!")
        sys.exit(0)
    else:
        print("Conversion failed!")
        sys.exit(1)

if __name__ == "__main__":
    main()
```

mikeller, 2025-05-27:
Nice work on the python import script!

Subsurface actually already supports the import of a couple of sqlite based formats, like this https://github.com/subsurface/subsurface/blob/master/core/import-shearwater.cpp

So it should probably be easy to add support for the Tusa format by swapping a few queries - is this something you want to look into?

mikeller, 2026-10-18:
Hi @0x0ACB,

Just a quick note of encouragement, the SQLite-based Shearwater importer (core/import-shearwater.cpp) is a solid, directly reusable pattern for this, and your converter script is basically a complete field-mapping spec already. No need to reply, just flagging we're watching for your PR whenever you get to it.