subsurface/subsurface #4536: [Bug]: |Issue importing log from divesoft liberty

mtfisher, 2025-06-15:
### Description

When attempting to import a dive log from a divesoft liberty, currently getting a `Unknown DC` error. I have attached a log file I am having issues witth

### Subsurface Version

6.0.5365

### Platform(s)

Windows

### Platform Version

Windows 10

### Divecomputer Model

Divesoft Liberty 2021

### Steps to Reproduce

1. Go to top menu
2. Click on `Import > Import log file`
3. Select a log exported from liberty

Attached a zip with a dive log in it

[L_Dive-01478656.zip](https://github.com/user-attachments/files/20747358/L_Dive-01478656.zip)

### Logs

```shell
INFO: validateGL(): created OpenGLContext.
INFO: validateGL(): obtained QOpenGLFunctions.
INFO: validateGL(): detected OpenGL version 4.6.
INFO: loading en-GB translations
QObject::connect(QQuickWindow, QDeclarativeGeoMap_QML_6): invalid nullptr parameter
INFO: DivePixmaps DPR: 1.000000 metrics: 24 16 sz_bigger: 40
INFO: returning cloud URL https://ssrf-cloud-eu.subsurface-divelog.org/git/michael.fisher86@gmail.com[michael.fisher86@gmail.com]
INFO: loading dive data from: https://ssrf-cloud-eu.subsurface-divelog.org/git/michael.fisher86@gmail.com[michael.fisher86@gmail.com]
Cloud storage: checking connection to cloud server https://ssrf-cloud-eu.subsurface-divelog.org/git/michael.fisher86@gmail.com
INFO: Checking cloud connection...
INFO: IP used for cloud server access 88.97.224.138
Cloud storage: successfully checked connection to cloud server
INFO: git storage: Synchronising data file
INFO: git storage: update local repo

INFO: git storage: sync with remote https://ssrf-cloud-eu.subsurface-divelog.org/git/michael.fisher86@gmail.com[michael.fisher86@gmail.com]

INFO: git storage: Sync with cloud storage
INFO: git storage: delete proxy setting

Cloud storage: checking connection to cloud server https://ssrf-cloud-eu.subsurface-divelog.org/git/michael.fisher86@gmail.com
INFO: Checking cloud connection...
Cloud storage: successfully checked connection to cloud server
INFO: git storage: fetch remote https://ssrf-cloud-eu.subsurface-divelog.org/git/michael.fisher86@gmail.com

INFO: git storage: Successful cloud connection, fetch remote
INFO: git storage: certificate callback for host ssrf-cloud-eu.subsurface-divelog.org with validity 1

INFO: git storage: certificate callback for host ssrf-cloud-eu.subsurface-divelog.org with validity 1

INFO: git storage: check remote status

INFO: git storage: try to update

INFO: git storage: Done syncing with cloud storage
INFO: git storage: Load dives from local cache
ERROR: Dive: 262, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 263, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 264, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 265, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 266, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 267, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 268, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 269, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 270, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 271, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 272, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 273, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 280, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 281, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 285, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 286, dive computer: DLF import: CCR dive, but no diluent cylinder found. Please add or select the correct cylinder use.
ERROR: Dive: 287, dive computer
```

mtfisher, 2025-08-04:
whats the progress on fixes?

mikeller, 2025-08-05:
#4538 needs a bit more work to include different Divesoft dive computer models, and I got the information from Divesoft to cover the work that is needed.
Unfortunately I am busy with other stuff at the moment.

glance-, 2025-08-06:
#4510 solves this and is ready to be merged.

mikeller, 2025-08-06:
#4538 solves all problems reported by Subsurface users with Divesoft dive computers, and it provides more relevant information to users compared to #4510, so if we all agree that getting this fixed now is the right approach #4538 is ready to be merged.

PR #4538 (2025-08-30): Divesoft: Fix the Log File Import and Add New Liberty Model.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [x] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fix the import of Divesoft log files to default to the Freedom model if
an unknown model identifier is encountered.
Also add support for the 'Liberty 2021' model identifier.

### Changes made:

1. fix the import of Divesoft log files to default to the Freedom model if an unknown model identifier is encountered;
2 [... cut]
commit a094c64d (2025-08-30): Divesoft: Fix the Log File Import and Add New Liberty Model.

Fix the import of Divesoft log files to default to the Freedom model if
an unknown model identifier is encountered.
Also add support for the 'Liberty 2021' model identifier, V1 log files, and factory test dives.

Fixes #4536.
closed 2025-08-30 by mikeller