subsurface/subsurface #4537: [Bug]: Unable to import dives from Perdix 2 firmware 100

teotwaki, 2025-06-15:
### Description

I recently upgraded my Perdix 2 to the latest firmware and since then I am unable to download dives through Bluetooth (the only connection possible to this dive computer).

I have reproduced the same behaviour on both iOS and macOS. After Subsurface attempts to download the dive logs, Subsurface generates a "Error Dive data import error" error (desktop version only). The dive computer is left in a semi-invalid "Wait Cmnd" state, where I then have to cancel the bluetooth mode and re-enable it, otherwise all operations time out.

I have enabled the creation of a libdivecomputer logfile (see below). I was unable to save a dumpfile.

### Subsurface Version

6.0.5365

### Platform(s)

macOS

### Platform Version

Sequoia 15.5, M1 Pro

### Divecomputer Model

Shearwater Perdix 2

### Steps to Reproduce

1. Open subsurface
2. Try to synchronise dive logs

### Logs

```shell
Subsurface: v6.0.5365.0, built with libdivecomputer v0.9.0-devel-Subsurface-NG (d2874be4fcb58385a62b335b7132aa17c4294925)
[46.105596] INFO: Open: transport=32
[46.105623] INFO: Configure: baudrate=115200, databits=8, parity=0, stopbits=0, flowcontrol=0
[46.105627] INFO: Timeout: value=12000
[46.105631] INFO: Sleep: value=300
[46.408319] INFO: Purge: direction=3
[46.408519] INFO: Write: size=10, data=0100FF010400228010C0
[46.462770] INFO: Read: size=18, data=010001FF0C006280104135454333324630C0
[46.462831] INFO: Write: size=10, data=0100FF010400228011C0
[46.524426] INFO: Read: size=22, data=010001FF10006280115631303020436C6173736963C0
[46.524435] ERROR: Unexpected packet size (12 bytes). [in src/shearwater_common.c:553 (shearwater_common_rdbi)]
[46.524445] ERROR: Failed to read the firmware version. [in src/shearwater_petrel.c:181 (shearwater_petrel_device_foreach)]
[46.524505] INFO: Write: size=11, data=0100FF0105002E902000C0
```

teotwaki, 2025-06-16:
It appears this is coming from libdivecomputer: 

[shearwater_petrel.c](https://github.com/subsurface/libdc/blob/4a20e1e86b6644ccd74c48e856ea66d78ee13dbc/src/shearwater_petrel.c#L179C1-L185C3)

```c
	// Read the firmware version.
	unsigned char rsp_firmware[11] = {0};
	rc = shearwater_common_rdbi (&device->base, ID_FIRMWARE, rsp_firmware, sizeof(rsp_firmware));
	if (rc != DC_STATUS_SUCCESS) {
		ERROR (abstract->context, "Failed to read the firmware version.");
		return rc;
	}
```

[shearwater_common.c](https://github.com/subsurface/libdc/blob/Subsurface-DS9/src/shearwater_common.c#L523)

```c
dc_status_t
shearwater_common_rdbi (shearwater_common_device_t *device, unsigned int id, unsigned char data[], unsigned int size)
{
        // ...
	unsigned int length = n - 3;

	if (length != size) {
		ERROR (abstract->context, "Unexpected packet size (%u bytes).", length);
		return DC_STATUS_PROTOCOL;
	}
```

So it would appear that the Perdix 2 in firmware v100 returns 12 bytes, not 11. Firmware v99 introduced Avelo support, so maybe they changed the response?

What would be the best way to provide the missing information, and how should we implement this change in behaviour depending on version level?

BlackFire01, 2025-06-18:
Same issue on Windows, completely unable to continue my divelog...

joshjohanning, 2025-06-19:
Same issue on iOS with a perdix 2 on firmware 100.

jefdriesen, 2025-06-19:
Yes, I already have a fix available and I'm waiting on the feedback from a tester.

It's a small bug in libdivecomputer. The firmware version is reported as the string "V100 Classic". Because the version jumped to V100, this string became one digit longer than previous versions. This causes an error because the code expects a fixed length (11 instead of 12 bytes). I have updated the code to support a variable length.

There is no need to contact Shearwater, but since you already did, let me know if there is updated documentation available.

BlackFire01, 2025-06-19:
Wow, that sure sounds like a Year 2000 problem! at least libdivecomputer will not break any more for 899 more SW firmware updates... 😄

BlackFire01, 2025-06-19:
Thanks! How do I incorporate the fix into my current Subsurface build on Windows?

mikeller, 2025-06-20:
A fix is available from https://subsurface-divelog.org/latest-release.

teotwaki, 2025-06-23:
Many thanks @jefdriesen and @mikeller for the prompt fix and release. Looking forward to test tonight.

joshjohanning, 2025-06-23:
@jefdriesen and @mikeller- Just tested this on macOS, the import worked 😄 

One thing I noticed was that the firmware version of 100 doesn't seem to get translated properly though in the dive log/file. 

[image]

The prior version had 99, and now for version 100 it has `a0`. 
[image]

joshjohanning, 2025-06-23:
It does say firmware version 100 in the logs! Attaching.  Let me know if you need anything else @jefdriesen 

[subsurface logfile.log](https://github.com/user-attachments/files/20872345/subsurface.logfile.log)

```
Event: model=0 (0x00000000), firmware=100 (0x00000064)
```

mikeller, 2025-06-23:
It looks like the 'firmware version' field in the actual log data (i.e. not received by interrogating the dive computer explicitly) is nibble-formatted, and we are displaying it as a hex number:

https://github.com/subsurface/libdc/blob/87a78d93e2fe3bc10d9840eade0c8f9572a169b2/src/shearwater_predator_parser.c#L873

joshjohanning, 2025-06-24:
> It looks like the 'firmware version' field in the actual log data (i.e. not received by interrogating the dive computer explicitly) is nibble-formatted, and we are displaying it as a hex number: [...]

so `99` gets brought in as 2 digits - but then 100 somehow gets `a0`?   To clarify, `< 100` firmware versions were being displayed correctly.

[image]

mikeller, 2025-06-24:
> [@mikeller](https://github.com/mikeller) so `99` gets brought in as 2 digits - but then 100 somehow gets `a0`? To clarify, `< 100` firmware versions were being displayed correctly.

Yes - we are actually printing this as a hex number, so 99 is encoded as `0x99` (153 decimal, or '1001 (9) 1001 (9)' in binary, hence the 'nibble' encoding) '100' is then '1010 (10) 0000' in binary, 160 decimal, or `0xa0` hexadecimal.

mikeller, 2025-06-24:
> Would something like this solve it? (not a `C` expert by any means):
> 
> -       dc_field_add_string_fmt(&parser->cache, "FW Version", "%2x", data[19]);
> +	// Special case for firmware versions
> +	// Firmware versions are stored in a modified Binary-Coded Decimal format:
> +	// - Versions < 100 are stored directly (e.g., 99 stored as 0x63)
> +	// - Versions >= 100 use a special encoding where the high nibble is 0xA + the tens digit - 10
> +	//   and the low nibble is the ones digit (e.g., 100 stored as 0xA0, 101 as 0xA1, etc.)
> +	if (data[19] >= 0xA0) {
> +		dc_field_add_string_fmt(&parser->cache, "FW Version", "%u", data[19] - 0xA0 + 100);
> +	} else {
> +		dc_field_add_string_fmt(&parser->cache, "FW Version", "%2x", data[19]);
> +	}
> In: [subsurface/libdc@`87a78d9`/src/shearwater_predator_parser.c#L873](https://github.com/subsurface/libdc/blob/87a78d93e2fe3bc10d9840eade0c8f9572a169b2/src/shearwater_predator_parser.c#L873)
> 
> Trying to account for the `< 100` and `>= 100` cases at the same time.

First of all I have asked Shearwater support to clarify how this firmware version is encoded - no point adding a fix for version 100 if we don't know if it will work for 101 or not.

Second, I'd probably rather do this by nibble - so something like `dc_field_add_string_fmt(&parser->cache, "FW Version", "%u%u", (data[19] & 0xF0) >> 4,  data[19] & 0x0F)`.

joshjohanning, 2025-06-24:
> First of all I have asked Shearwater support to clarify how this firmware version is encoded - no point adding a fix for version 100 if we don't know if it will work for 101 or not.
>
> Second, I'd [...]

Makes sense to me! Thank you 🙇

jefdriesen, 2025-06-24:
My guess: it's originally a BCD encoded number. This can normally represent numbers up to 99. However, when allowing the full range of the first digit (e.g. 0-F instead of just 0-9) this limit increases to 159. Still worse than the 255 of a plain 8 bit number, but backwards compatibility is preserved.

`dc_field_add_string_fmt(&parser->cache, "FW Version", "%u", bcd2dec(data[19])`

mikeller, 2025-06-24:
> My guess: it's originally a BCD encoded number.

Likely to be correct.

> This can normally represent numbers up to 99. However, when allowing the full range of the first digit (e.g. 0-F instead of just 0-9) this limit increases to 159. 

My guess is they will start using '1010' binary and so on for the second nibble when they get to 160 - which will allow them to get to 165. 🤪

joshjohanning, 2025-07-28:
@jefdriesen @mikeller Circling back to the past couple of messages. Do we think it's possible to correct the `a0` to `100` for the firmware version in the Extra Info tab?  I can create it as a separate issue if we want to track there.  Thanks 😄