subsurface/subsurface #4553: [Feature Request]: Native Apple Silicon (arm64) build of the macOS App

wchmb, 2025-07-11:
### Problem Statement

Hi!

First all, thank you for your work on this project, much appreciated!

I noticed that the [latest v6.0 .dmg release](https://subsurface-divelog.org/downloads/Subsurface-6.0.5404-CICD-release.dmg) is built for Intel architecture (x86_64) only. When running the app on Apple Silicon, this requires Rosetta for translation, which is... not desirable

Is there any plan or roadmap for releasing a native Apple Silicon (arm64) or Universal build? 

Happy to help with testing if needed

### Description

I inspected the binary to confirm it's Intel-only:
```
lipo -archs /Volumes/Subsurface-6.0.5404/Subsurface.app/Contents/MacOS/Subsurface
#Output: x86_64
```

atdotde, 2025-07-11:
Subsurface heavily relies on the Qt libraries providing the user interface. More specifically, Subsurface uses version 5 of Qt which is not compatible with Qt version 6. There are attempts on the way to make Subsurface work with Qt6, but those are far from complete (Qt has discontinued various parts in version 6 that Subsurface uses for example for printing). On the other hand, Apple Silicon is only supported by Qt6. So it's not a simple setting in the compile options unfortunately. At the moment it is therefore preferable to use version that involves Rosetta and is thus a bit slower  to a version where several major features are missing.

At least, this was the state of affairs a while ago. I might have missed more recent developments but I don't think so.

mikeller, 2025-07-11:
@wchmb: #4302 is the latest attempt at moving to Qt6, with some discussion on the issues encountered along the way. If you are keen to help this will be much appreciated.

wchmb, 2025-07-11:
@mikeller so https://github.com/subsurface/subsurface/tree/Egbertdepauw-qt6devel is the most advanced branch in this regard?
It is not easy for newcomers like me to follow the rationale behind this repository [branching](https://github.com/subsurface/subsurface/branches/all)

From https://github.com/subsurface/subsurface/pull/4302#issuecomment-2320121521
I'm agree with you that printing and an integrated manual are totally dispensable for many.

In the meantime, I've tried [v5.0.10-Qt6-universal release](https://subsurface-divelog.org/downloads/Subsurface-5.0.10-qt6-universal.dmg) and it works pretty fine...

mikeller, 2026-10-17:
Hi @wchmb,

This is the same underlying situation as #4946, which we've already answered: native Apple Silicon (arm64) builds exist in our Qt6 development/CICD release stream, for example Subsurface-arm64-qt6-6.0.5621-CICD-release.dmg. These aren't yet the default macOS build pending more testing and feature parity, but they're actively maintained and you can install them side-by-side with the Intel build for testing. Current dev builds are always at https://subsurface-divelog.org/latest-release/.

Closing this as addressed since arm64 builds are available today. Please reopen if you hit problems specific to Apple Silicon once you've tried one.

closed 2026-10-17 by mikeller