subsurface/subsurface #4557: [Bug]: ERROR: Unable to open /dev/ttyUSB0 Seac (Screen)

walkjivefly, 2025-07-16:
### Description

Attempting to download dive logs from Seac Screen results in a popup saying Error: Unable to open /dev/ttyUSB0 Seac (Screen)

Running the latest AppImage on MX-Linux 23.

This has worked in the past on a different PC (running older AppImage and MX-Linux 21). It looks like a permissions problem so I tried running it as root and got the same error.

### Subsurface Version

6.0.5404-CICD-release

### Platform(s)

Linux

### Platform Version

MX-Linux 23

### Divecomputer Model

Seac Screen

### Steps to Reproduce

1. Open subsurface
2. Import, import from dive computer
3. Set vendor=Seac, dive computer=Screen, device=/dev/ttyUSB0, hit Download
4. It says Connecting, thinks for a few seconds and then pops up the error box.

### Logs

```shell
INFO: Starting download from SERIAL
INFO: downloading only new dives
[1.232443] ERROR: Failed to receive the packet start byte. [in src/seac_screen.c:196 (seac_screen_receive)]
[1.232452] ERROR: Failed to receive the response. [in src/seac_screen.c:282 (seac_screen_packet)]
[2.339111] ERROR: Failed to receive the packet start byte. [in src/seac_screen.c:196 (seac_screen_receive)]
[2.339122] ERROR: Failed to receive the response. [in src/seac_screen.c:282 (seac_screen_packet)]
[3.446227] ERROR: Failed to receive the packet start byte. [in src/seac_screen.c:196 (seac_screen_receive)]
[3.446236] ERROR: Failed to receive the response. [in src/seac_screen.c:282 (seac_screen_packet)]
[4.552658] ERROR: Failed to receive the packet start byte. [in src/seac_screen.c:196 (seac_screen_receive)]
[4.552668] ERROR: Failed to receive the response. [in src/seac_screen.c:282 (seac_screen_packet)]
[5.660206] ERROR: Failed to receive the packet start byte. [in src/seac_screen.c:196 (seac_screen_receive)]
[5.660216] ERROR: Failed to receive the response. [in src/seac_screen.c:282 (seac_screen_packet)]
[5.660219] ERROR: Failed to read the hardware info. [in src/seac_screen.c:360 (seac_screen_device_open)]
INFO: INFO: dc_device_open error value of -7
INFO: Finishing download thread: Unable to open /dev/ttyUSB0 Seac (Screen)
```

walkjivefly, 2025-07-17:
> First of all: Don't run subsurface as root. That creates all kinds of problems because from that moment, some essential files are no longer accessible to the normal user. In fact, Subsurface should refuse to start when invoked with superuser rights.
> 
Attempting to run as root is a well known technique for quickly establishing if a problem really is a permissions one. In this case, it seems not.

> Better solve the actual problem: Does the device /dev/ttyUSB0 actually exist? if not, it might have a different name or there is a USB problem. What does lsusb say? Who owns that device? Maybe some group called modem? Then add your user to that group. To become effective, you will have to log out and then log in again.

/dev/ttyUSB0 does exist and corresponds to the USB/UART adapter connected to the dive computer. The device is owned by root, group is dialout and my user is in the dialout group.

> 
> Maybe some other process grabbed that device? Here lsof might help.
Nothing else has grabbed the device. While the download is failing to occur lsof shows Subsurface is talking to it.

I fired up the old PC having forgotten I'd upgraded it to MX-23 since last running Subsurface. But the old PC still had subsurface 5.0.9 installed and running that version on both machines results in the same error.  So can't rule out the OS version as cause of the problem (although it seems unlikely given I'm running an AppImage in both cases). But it's almost certainly not the cable or USB port(s) causing it. 

Given a bit more time and effort I can get MX-21 running again on the old machine and see if the problem still repros with either Subsurface version....

mikeller, 2025-07-17:
> I fired up the old PC having forgotten I'd upgraded it to MX-23 since last running Subsurface. But the old PC still had subsurface 5.0.9 installed and running that version on both machines results [...]

As @atdotde says, finding out if something in MX-23 is grabbing the port might be a good next step `lsof /dev/ttyUSB0` will be telling you more.

closed 2025-07-24 by walkjivefly