subsurface/subsurface #4607: [Feature Request]: location filtering should include a dive site's country

sothawo, 2025-09-26:
### Description

When filtering for a location I can specify the location's name, but not the country. I'd like to be able to use the country for a location as well.

I am monitoring german lakes and would like filter for dives "in Germany". Although the country is a separate property in the dive site definition, it is not used in filtering.

I'll submit a PR to add this functoinality.

mikeller, 2025-09-27:
> ### Alternative Solutions

You can just use the already existing full text search functionality for this:

[image]

Your proposal seems to make the 'location' filter less specific, while not providing additional value over what is already supported.

sothawo, 2025-09-27:
Didn't check the fulltext. Yes that's working for my case (I currently analyse water temperatures in my dives in lakes in Germany and need to exclude some dive from Spain and France there).

But using the full text search, I might get some entries I don't want. If I had a dive in Australia, and in my notes I might have "nice trip with a couple from Germany". 

Another thing that cannot be done with the full text approach is negating the condition, like in the following screenshot

[image]

Although that's not my use case.

Using the full text approach works for me, although having the restriction only on the location seems more precise.

mikeller, 2025-09-27:
Makes sense.

PR #4608 (2025-09-27): consider a dive site's country in the location filter

### Describe the pull request:

- [ ] Bug fix
- [ ] Functional change
- [x ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Filtering for a dive site's location now uses the country as well, so it's possible now to filter  for countries as well.
### Changes made:

### Related issues:

Fixes #4607 

### Additional information:

### Documentation change:
closed 2025-09-27 by mikeller